fritzing/fritzing-app #4029: Duplicate entries for trace width

KjellMorgenstern, 2023-05-17:
## Current Behaviour
Duplicate entries for custom wire thickness:
[image]

**Build:**
Fritzing 1.0.0

**Operating System:**
Linux

**Steps to reproduce:**
1. Load Arduino No FTDI example
2. Go to PCB
3. Select thin wires near the XTAL
4. Open the "width" selection in the Inspector.

It looks like the values get even duplicated when opening the selection mutltiple times.

## Expected Behaviour
No duplicates

vanepp, 2023-05-19:
For me on Windows 10 it appears to be working normally (unless I am not understanding how to reproduce it!) 

Version 1.0.0 
(rcCD-4028-0-eb6a9551 2023-05-11) 64 [Qt 6.4.3]

[image]

Is the 24 mil selection appearing multiple times in the Inspector window for you? It doesn't appear to be for me. 

[image]

everything appears normal as far as I can see.

KjellMorgenstern, 2023-05-19:
It might be related to the exact value of the width.
I didn't set the thin width directly, but triggered another bug, that causes the width to be "11" ( Go to Breadboard, delete XTAL, press undo, go back to PCB -> trace width was not restored correctly)

vanepp, 2023-05-19:
Yes that works better (although still slightly differently for me!)

[image]

then delete the crystal and redo it and the trace size changes to 8 mil

[image]

while it looks to be selecting 8mil, it indeed has the 11 as an option in the menu and will let me select it as the width (although I think that is normal operation if you manually set a trace width!) 

[image]