fritzing/fritzing-app #4037: Mouse pointer strange behaviour

Motomanual, 2023-06-26:
Hello, just installed version 1.0.0-1457Win64 but found a strange behaviour of the mouse pointer when focused on a component on breadboard and drawing sections (pointer looks like two orizontal lines).
Already sent a mail to 'support@fritzing.org' with a short video showing the issue, that was not present on previous versions and make complex connect wires to components.
Additionally in the Code section the drop down menu used to select platform/card/port, words appear in black color on dark gray background (almost unreadable).
The mouse problem appear on a specific laptop below ireported and just in the indicated sections, color problem appeared on all pc I have tested.
O.S. - Windows 10 V19045.3086
PC Lenovo Ideapad Ryzen 7
Thanks and best regards

Motomanual, 2023-06-26:
https://github.com/fritzing/fritzing-app/assets/137813515/13baee7c-2e8e-45e0-b2f6-30d5cf43f2b5

KjellMorgenstern, 2023-06-29:
Can you check the integrity of Fritzing? Reachable by right clicking, ( https://chat.openai.com/share/ce4247fa-e95c-4188-b612-21fc64c1f7a1 ) . Certificate should be to Fritzing GmbH.

Were there any error messages when installing on the Ideapad?

Motomanual, 2023-06-29:
Hello, thanks for Your answer!
I checked the integrity as requested and the certificate is correct (I download the file from the link supplied by Fritzing several times).
During installation any error message appeared (checked several times), previous versions don't have this problem.
Thanks and best regards.

failiz, 2023-07-17:
I have seen this bug in a few computers from my students (all running 1.0.0).

KjellMorgenstern, 2023-07-17:
Unfortunately I have never seen it yet. Were all the students using Windows? And can you check if it was Windows 10, maybe even build 19045 like in the first post?

failiz, 2023-07-17:
I think all of them were on Windows. I took some pictures of one of them. I will try to find more tomorrow.
[image]
[image]
[image]

failiz, 2023-07-18:
Another computer with the same problem.
[image]
[image]
[image]

Let me know if you need more info.

KjellMorgenstern, 2023-07-18:
Are there any Windows PC where it does work? 

One thing suspicious is that the cursors are encoded as binary bitmap
`make_wire.bmp: PC bitmap, Windows 3.x format, 32 x 32 x 1, image size 130, resolution 2834 x 2834 px/m, cbSize 192, bits offset 62`
Maybe that is to outdated for some graphics drivers, but just pure speculation here.

Motomanual, 2023-07-18:
Hello, in my case there are other 3 (older) laptops with same Windows version that don't have any problem.
In the newest laptop the problem appear just with Fritzing version 1.0.0, older versions (included 0.9.10 that's Im actually using) dont have any problem and suppose the cursor bitmap is the same...
What about the QT version?

failiz, 2023-07-18:
> Are there any Windows PC where it does work?

I have 23 students and around half of them (or 40%) are MacOS users. From the around 12 or 14 Windows users, I have seen this bug in two or maybe three computers. In the other ones, it works correctly.

jfmorgenstern, 2023-07-18:
I was able to reproduce the bug by adjusting the display scaling to a non-integer scaling factor of 100%, such as 175%.

[image]

Tested on Windows 11 
[image]

Motomanual, 2023-07-18:
You're right, the suggested scaling setting of my monitor is 125% but changing to 100% the problem disappear, unfortunately with this scaling setting there are reading problem with other programs..
An additional issue that forgot to report is that, after installation, the resolution of version 0.9.10 and 1.0.0 is quite different as shown on below pictures (scaling 125%, 1.0.0 is bigger).
[image]
[image]

jfmorgenstern, 2023-07-18:
This could be related to the switch to QT6.

As a workaround until we fix this, you could try to change the high DPI settings in the compatibility tab.

1. Right click on Fritzing shortcut
2. Select Properties
3. Select Compatibility tab
[image]
4. Set High DPI scaling override to "Sytem" or "System Enhanced"
[image]

This fixes the cursor for me.

Motomanual, 2023-07-18:
With this setting the cursor probem is fixed also for me, the resolution of this version remain bigger compared to 0.9.10 but now is possible to use it.
Thanks for your support!

Motomanual, 2023-07-18:
Hello, in my first post I mention another problem 'Additionally in the Code section the drop down menu used to select platform/card/port, words appear in black color on dark gray background (almost unreadable).' and don't know if is necessary open a new issue, here a picture showing it.
[image]

KjellMorgenstern, 2023-07-18:
Of course it is, split the problems. Thats what the issues are for.

KjellMorgenstern, 2023-07-21:
I added it: https://fritzing.org/download/known-issues/

Spinky1001, 2023-08-30:
I started using Fritzing yesterday and I also  have this problem with the cursor.
Tried all the suggestions above but still not working properly.
Not sure what last comment means - how do you switch the cursor from Bitmap to Pixmap?
Any further ideas?
Windows 10 home edition.

closed 2023-11-03 by KjellMorgenstern