fritzing/fritzing-app #4038: Arduino Code support.

ouTPhaze, 2023-06-29:
## Problem
Simulating any Arduino Application is currently impossible.

## Proposed Solution
Allow the capability to simulate Arduino Code within the program, this could do wonders when prototyping.

For an example of Arduino Code simulation in action, check out: https://tinkercad.com

Thank you for your time.

KjellMorgenstern, 2023-06-29:
If you want this request to be taken serious, a way more detailed ticket is needed (my guess chances are dim, but without a detailed look, we won't know)

While is a very common request, does it really solve a problem? "Sketching in hardware" is a real thing. Do you really learn electronics when pressing a "Play" button?

How many people would want to use that simulator?

By our current calculation, this would be quite expensive to develop. How many free Arduinos could we give out to everyone who wanted a simulator? 

How accurate does the simulator need to be?

Do you just want to see some movement and blinking on the screen? Or do you want the exact same problems, that a real circuit could show?

If you use the simulator, do you want to diagnostics about your Arduino code running? Or your circuit?
Or have you already checked these, and want to know to check the integration of both?

How much effort would that be, what is your estimate? (Tip: We won't know until we have better requirements)

Are there existing techs we can use, and how?

Please add more questions like that, just getting started here ;-) Answers not yet needed.

Sasmeet1997, 2023-08-04:
An Arduino  simulator would be wonderful.  Was thinking about teaching kids  about electronics and this is the perfect software.

KjellMorgenstern, 2023-09-17:
This looks like one of the most requested features, but we need numbers, as it would be quite an investment. Please use the "Thumbs up" feature of github :+1: