fritzing/fritzing-app #4040: Hint: add components

AntonioBerna, 2023-07-06:
## Problem
A few days ago, the new Arduino Minima and Arduino Wifi (integrated with the ESP8266) were unveiled, and I would like to have the ability to create assembly diagrams that are graphically compatible with these new microcontrollers.

## Proposed Solution
It would be very convenient to add the new Arduino Minima and Arduino Wifi microcontrollers to the Arduino section of Fritzing, and also if someone wanted to work with the ESP32 or ESP8266 it would be convenient to have the ability to create assembly diagrams with Fritzing.

KjellMorgenstern, 2023-07-06:
Duplicate

closed 2023-07-06 by KjellMorgenstern

AntonioBerna, 2023-07-07:
I can't find the duplicate, could you send me the link?

KjellMorgenstern, 2023-07-07:
Fritzing parts are managed in the fritzing-parts repo. Also, please stick to one part per issue, unless it is a systematic bug or similar that affects many at the same time.