fritzing/fritzing-app #4044: Compiling issue

bsbhro, 2023-07-19:
## Current Behaviour
Hello, my attempt to compile ends with this error. I do not have c++ skills.
---snip
src/sketch/sketchwidget.cpp: In member function ‘virtual long int SketchWidget::setUpSwap(SwapThing&, bool)’:
src/sketch/sketchwidget.cpp:5962:12: error: ‘partLabel’ was not declared in this scope; did you mean ‘PartLabel’?
 5962 |         if(partLabel != nullptr) {
      |            ^~~~~~~~~
      |            PartLabel
---/snip
**Build:**
latest code
**Operating System:**
Ubuntu 23.04
**Steps to reproduce:**
- Step 1: git pull, qmake
- Step 2: make

Thank you for this software and any help.

DJManas, 2023-07-25:
Workaround, until solved.

Renamed partLabel to oldPartLabel on this line. It compiles, but dunno if it is needed (don't have time to get into whole code). App seems to be working.

supercairos, 2023-07-31:
I have the same issue.

bsbhro, 2023-08-03:
I cloned new and the error is gone. Thank you very much!

closed 2023-09-10 by KjellMorgenstern