fritzing/fritzing-app #4045: Low Code menu readability

Motomanual, 2023-07-19:
## Current Behaviour
In the CODE section the drop down menu used to select platform/card/port, options appear in black color on dark gray background (almost unreadable).
**Build:**
<Version 1.0.0 (rcCD-4157-0-6b14cfe 2023-06-02) 64 (Qt6.4.3)

**Operating System:**
<Windows 10 V19045.3086>

[image]
Problem appear with dark and light theme and was not present on previous versions.

jfmorgenstern, 2023-07-20:
I could reproduce this on Windows 11.

Motomanual, 2023-07-21:
Are there any settings to improve the readability?

KjellMorgenstern, 2024-01-10:
It should look like this in dark mode, tested with 1.0.2, but on Linux:

[image]

closed 2024-06-26 by KjellMorgenstern