fritzing/fritzing-app #4048: DRC does not check for unrouted connections

failiz, 2023-07-26:
## Current Behaviour
Create a PCB by designing some schematics and change to PCB view. Run the DRC test. The DRC message states: "Your sketch is ready for production: There are no connectors or traces that overlap or are too close to each other". That message is misleading. If you order the PCB, it will not work as there are no traces.

[image]

**Build:**
Fritzing 1.0.0

**Steps to reproduce:**
- Change to SCH view
- Add two resistors
- Connect both the ends of both resistors with a wire
- Change to PCB view
- Run the DRC test

## Expected Behaviour
The DRC test should warn you that there unrouted connections.

KjellMorgenstern, 2023-08-23:
Fritzing shows the number of unrouted connections in the bottom bar, and highlights them when clicking there.

In that case, the dialog could say:

tr("There are still unrouted nets. Click the status in the bottom bar to highlight them.")

A more advanced fix could show the DRC Dialog, lists the nets and highlight them.

KjellMorgenstern, 2023-08-23:
Rating as bug, because "Ready for produciton" is misleading.