fritzing/fritzing-app #4051: Making adjustments to the Y position of a part introduces an offset to the X co-ordinate instead.

GOTO-GOSUB, 2023-08-12:
I have just started to make a simple PCB layout that uses 2 rows of headers. I've started with these and once they are placed accurately I will lock them in place. However in trying to place these headers I cannot get the Y position to place accurately. This is not the small offset (rounding error ?) that seems to be common to Fritzing, rather that if I make a change to the Y position by typing the desired numbers in, the X position changes instead.

Entering a value into the Y location should not affect the X co-ordinate.

If I manually drag the part then the XY co-ordinates change as expected but I cannot place the part accurately (I have turned off all the "snap to grid" type options that I can find.

### My version of Fritzing and my operating system:

1.0.0 / Windows 10

A copy of the affected file and screenshot is attached.

[image]

[Bare Expansion Board.zip](https://github.com/fritzing/fritzing-app/files/12327260/Bare.Expansion.Board.zip)

vanepp, 2023-08-12:
One thing to try is to zoom pcb vie out to maximum. There was (I'm not sure if it is fixed in 1.0.0) a roundoff error related to image scale that made the positioning inaccurate. Zooming all the way out before making the coordinate change helps.

GOTO-GOSUB, 2023-08-13:
Thank you for the suggestion. I've tried that and I still cannot place the headers at a specific point (eg a Y value of 17mm). Plus we've still got the issue where a change to the Y co-ordinate affects the X.

KjellMorgenstern, 2023-08-13:
Entering coordinates only works as expected if rotation is 0. If any rotation is applied, coordinates get messed up.

KjellMorgenstern, 2023-08-13:
This is related to the fix fa4e7b9 . 

Behavior before the commit: The inspector would show the same position for these two Rotations of R1.
Position is marked with the red cross:
[image]

[image]

When a part is rotated, the Inspector would show the wrong position. 

After the fix, the position shown is always the top left corner of the item.

However, now changes applied through the Inspector are done in the coordinate system of the item, which is unexpected.

GOTO-GOSUB, 2023-08-15:
Thank you to everyone who spent time on this, it is appreciated.

closed 2023-11-03 by KjellMorgenstern