fritzing/fritzing-app #4052: Fritzing 1.0.0 crashes constantly (MacOS 12.6.7)

z-l-p, 2023-08-15:
## Current Behaviour
Just paid and downloaded the latest version but it's not usable. It crashes on every save or load.
**Build:**
Version 1.0.0 (rcCD-4157-0-6b14cfe9 2023-06-02) Cocoa [Qt 6.4.3]

**Operating System:**
MacOS Monterey 12.6.7 (Intel)

**Steps to reproduce:**
Open Fritzing
File / Open or Save / select any file (grant OS permissions when asked)
The Qt load or save progress bar will appear but then Fritzing crashes.
**Crash Log 1:
Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   CoreFoundation                	    0x7ff812ccdf13 __CFStringCreateImmutableFunnel3 + 877
1   CoreFoundation                	    0x7ff812cdbb1f CFStringCreateWithCharacters + 38
2   Foundation                    	    0x7ff813b721a5 +[NSString stringWithCharacters:length:] + 40
3   libqcocoa.dylib               	       0x1059b3b35 0x105987000 + 183093
4   libqcocoa.dylib               	       0x1059ad6c7 0x105987000 + 157383
5   libqcocoa.dylib               	       0x1059b03e2 0x105987000 + 168930
6   libqcocoa.dylib               	       0x1059af783 0x105987000 + 165763
7   QtCore                        	       0x1052d35aa 0x105216000 + 775594
8   QtGui                         	       0x103fad2c5 QGuiApplicationPrivate::processActivatedEvent(QWindowSystemInterfacePrivate::ActivatedWindowEvent*) + 901
9   QtGui                         	       0x104005238 0x103f25000 + 918072
10  QtCore                        	       0x105293894 QMetaMethod::invoke(QObject*, Qt::ConnectionType, QGenericReturnArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument) const + 1284
11  libqcocoa.dylib               	       0x1059d3e62 0x105987000 + 314978
12  Foundation                    	    0x7ff813b7d48d -[__NSObserver _doit:] + 287
13  CoreFoundation                	    0x7ff812d4073c __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 12
14  CoreFoundation                	    0x7ff812dddb32 ___CFXRegistrationPost_block_invoke + 49
15  CoreFoundation                	    0x7ff812dddab0 _CFXRegistrationPost + 496

**Crash Log 2:
Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib               	    0x7ff812b17417 objc_msgSend + 23
1   libqcocoa.dylib               	       0x10243c8a4 0x102410000 + 182436
2   libqcocoa.dylib               	       0x1024366c7 0x102410000 + 157383
3   libqcocoa.dylib               	       0x1024393e2 0x102410000 + 168930
4   libqcocoa.dylib               	       0x102438783 0x102410000 + 165763
5   QtCore                        	       0x1039e75aa 0x10392a000 + 775594
6   QtGui                         	       0x101bd42c5 QGuiApplicationPrivate::processActivatedEvent(QWindowSystemInterfacePrivate::ActivatedWindowEvent*) + 901
7   QtGui                         	       0x101c2c238 0x101b4c000 + 918072
8   QtCore                        	       0x1039a7894 QMetaMethod::invoke(QObject*, Qt::ConnectionType, QGenericReturnArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument) const + 1284
9   libqcocoa.dylib               	       0x10245ce62 0x102410000 + 314978
10  Foundation                    	    0x7ff813b7d48d -[__NSObserver _doit:] + 287
11  CoreFoundation                	    0x7ff812d4073c __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 12
12  CoreFoundation                	    0x7ff812dddb32 ___CFXRegistrationPost_block_invoke + 49
13  CoreFoundation                	    0x7ff812dddab0 _CFXRegistrationPost + 496
14  CoreFoundation                	    0x7ff812d123c8 _CFXNotificationPost + 735
15  Foundation                    	    0x7ff813b517ee -[NSNotificationCenter postNotificationName:object:userInfo:] + 82

**Crash Log 3:
Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib               	    0x7ff80f554417 objc_msgSend + 23
1   libqcocoa.dylib               	       0x108de98a4 0x108dbd000 + 182436
2   libqcocoa.dylib               	       0x108de36c7 0x108dbd000 + 157383
3   libqcocoa.dylib               	       0x108de63e2 0x108dbd000 + 168930
4   libqcocoa.dylib               	       0x108de5783 0x108dbd000 + 165763
5   QtCore                        	       0x10a3945aa 0x10a2d7000 + 775594
6   QtGui                         	       0x1085812c5 QGuiApplicationPrivate::processActivatedEvent(QWindowSystemInterfacePrivate::ActivatedWindowEvent*) + 901
7   QtGui                         	       0x1085d9238 0x1084f9000 + 918072
8   QtCore                        	       0x10a354894 QMetaMethod::invoke(QObject*, Qt::ConnectionType, QGenericReturnArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument) const + 1284
9   libqcocoa.dylib               	       0x108e09e62 0x108dbd000 + 314978
10  Foundation                    	    0x7ff8105ba48d -[__NSObserver _doit:] + 287
11  CoreFoundation                	    0x7ff80f77d73c __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 12
12  CoreFoundation                	    0x7ff80f81ab32 ___CFXRegistrationPost_block_invoke + 49
13  CoreFoundation                	    0x7ff80f81aab0 _CFXRegistrationPost + 496
14  CoreFoundation                	    0x7ff80f74f3c8 _CFXNotificationPost + 735
15  Foundation                    	    0x7ff81058e7ee -[NSNotificationCenter postNotificationName:object:userInfo:] + 82

(I hope these are useful. I can supply more logs if it helps.)

## Expected Behaviour
Files should load or save without crashing

KjellMorgenstern, 2023-08-17:
I tried to open and save files on macOS Montery, but it would have been to simple if it was crashing here, too.

Are you using network drives, like Apple iCloud to open or store Fritzing files? Or other temporary storages, like USB drives?
Can you open the examples (in the file menu, below the open and save section)?

Unfortunately, the crash report is not very helpful, since none of the calls contains Fritzing code, only Qt and other libraries. Does the report contain information about different threads (other then "Thread 0") ?

Also, Fritzing has a mechanism to intercept the crash, and show an error message before closing. Do you see such an error message, if yes, can you screenshot it?

KjellMorgenstern, 2023-09-10:
Not enough information available, can not reproduce the crash.

closed 2023-09-10 by KjellMorgenstern