fritzing/fritzing-app #4053: svgpp/svgpp.hpp not detected.

kaamburoow, 2023-08-20:
Hello, 
Today I decided to compile Fritzing on Linux(Ubuntu 22.04). I used QT Creator, QT 6.4.3, Boost 1.81, ngspice-40. When building it gives me this error: svgpp/svgpp.hpp: No such file or directory. I assumed that svgpp wasn't added in the docs, so I downloaded it. Looked in svgppdetect.pri that is searching for a folder named svgpp-1.3.0 in the folder, where fritzing-app, fritzing-parts, libgit2, boost_1_81_0, ngspice-40 and quazip_qt6 are located. Downloaded svgpp, extracted it and tried to build again. Again QT Creator gave me this error. Looked again in svgppdetect.pri and saw that it is including svgpp-1.3.0/include folder. I have that folder and svgpp/svgpp.hpp, but still the same error. Could anyone help me?

Virinas-code, 2023-08-26:
I've been trying to build Fritzing for hours now. The docs are really outdated. I switched to Qt6 because it uses some Qt6-only features. Good luck. Btw I think you have to install svgpp with make etc.

kaamburoow, 2023-08-27:
You have to use QT6, because the 1.0.0 change log states the project has switched to QT 6.4.3, Boost 1.81.0 and ngspice-40.

Amrzxc, 2023-08-30:
same problem using qt6 anyone find a solution ?

rob4226, 2023-09-07:
> same problem using qt6 anyone find a solution ?

I haven't seen it mentioned anywhere but you need to include the [SVG++ library 1.3.0](https://github.com/svgpp/svgpp) which you can download from [here](https://github.com/svgpp/svgpp/releases/tag/v1.3.0). Unzip/untar the source code and put it next to the other dependencies like boost/quazip/etc... The folder will be named `svgpp-1.3.0`.

rob4226, 2023-09-08:
> > > same problem using qt6 anyone find a solution ?
> > 
> > 
> > I haven't seen it mentioned anywhere but you need to include the [SVG++ library 1.3.0](https://github.com/svgpp/svgpp) which you can download from [here](https://github.com/svgpp/svgpp/releases/tag/v1.3.0). Unzip/untar the source code and put it next to the other dependencies like boost/quazip/etc... The folder will be named `svgpp-1.3.0`.
> 
> I did the same and it doesn't get included...

Try including its path like `"INCLUDEPATH+=A:\fritzing\svgpp-1.3.0\include"` (change the path to where ever it really is) in the build section in Qt Creator. I also had to add ng-spice in this way:

[image]

kaamburoow, 2023-09-08:
Hi, rob4226! This solved my issue, so I was able to build Fritzing. However when running it now displays an error telling me that zlib.dll was not found when I have it in C:\Program Files\zlib\bin. Have you ever encountered this error? If yes, may you help? Thank you!

kaamburoow, 2023-09-08:
I fixed the issue by myself. Explanation: I copied zlib.dll from C:\Program Files\zlib\bin to where the debug64 folder is. After that Fritzing ran normally.

closed 2023-09-08 by kaamburoow