fritzing/fritzing-app #4056: PNP (Pick-and-Place) file is not a common Format (XLS,XLSX,CSV, etc.)

brdane, 2023-08-28:
## Problem
PCB manufacturers like JLCPCB cannot accept the .xy file format that Fritzing exports.

## Proposed Solution
Implement exporting of the PNP file to XLS, XLSX and CSV... along with the existing .xy format

KjellMorgenstern, 2023-08-28:
The xy file is in csv format.

closed 2023-08-28 by KjellMorgenstern

brdane, 2023-08-28:
Even if JLCPCB is unable to load it?

KjellMorgenstern, 2023-08-28:
Why can't they load our CSV?

brdane, 2023-08-28:
I am not sure why. All I know is that I took the .xy file, changed the extension to CSV since that is what they accept, but it still does not upload for some reason. 

[image]

KjellMorgenstern, 2023-08-28:
From the screenshots, it looks like you tried to upload the bill of materials, not the pick and place file.

KjellMorgenstern, 2023-08-28:
From the error message I can see as much information as you. Maybe contact jlcpcb support about it?

brdane, 2023-08-28:
Contacted their support and they said that both BOM and PNP files are a different format/layout than what fritzing generates. In the screenshots below I have the BOM/PNP on one side and JLCPCB's example BOM/PNP that they accept. Labels are different, there seems to be a lot of information that they, in-particular, do not need.

[image]
[image]

brdane, 2023-08-28:
For the PNP screenshot, my version of Excel (for some reason) opens its incorrectly, but it does show correctly in LibreOffice Calc. Regardless, JLC can't upload it. Their support member even made modifications to it as well and asked me to upload, but it also did not work.

brdane, 2023-08-29:
Would it help your team if I uploaded JPC's example BOM/PNP files as well as other ones?