fritzing/fritzing-app #4057: Using wrong version of Boost. Boost 1.54

perottosson, 2023-08-28:
## Current Behaviour
Im trying to build Fritzing using the instructions here:
https://github.com/fritzing/fritzing-app/wiki/1.-Building-Fritzing

Here:
https://siytek.com/build-fritzing-windows/ 

And here:
https://www.youtube.com/watch?v=deyyxNkQPSM

They are basically the same and produces the same error
```
Checking for boost... no
Project MESSAGE: Boost 1.54 has a bug in a function that Fritzing uses, so download or install some other version
Project ERROR: Easiest to copy the Boost library to ..., so that you have .../boost_1_xx_0
```

How can I change version of Boost? Copying it into the folderstructure as below doesnt work! 

My folderstructure looks like this:
[image]

I am using:
**I am using**
libgit2-0.28.5
boost_1_80_0
and latest version on gitlab

**Operating System:**
Windows 10

**Steps to reproduce:**
- Step 1
- Step 2

perottosson, 2023-08-29:
using boost 1.81.0 solved it. However now I got another bug.
```

Project MESSAGE: Using fritzing ngspice detect script.
Project ERROR: ngspice not found in C:/Users/myUser/Desktop/Fritzing/source/ngspice-40
```

kaamburoow, 2023-09-08:
Hello, perottoson! This issue was solved in #4053. Please have a look in there. You will also need svgpp-1.3.0 and ngspice-40.

closed 2023-09-10 by KjellMorgenstern