fritzing/fritzing-app #4059: Export PCB design to 3D Model

brdane, 2023-09-01:
## Problem
Users don't have a way to work their PCB design into an enclosure, in the case that they are developing a 3D model of one.

## Proposed Solution
Be able to export the user's PCB design into a realistically-sized 3D model, preferably STL, that the user can load into a 3D-modelling software and accurately build and size an enclosure that perfectly can house the PCB.

brdane, 2023-09-12:
That's a great first step, and if there can be multiple model formats added on top of that to accommodate to a vast amount of users with different preferences of software, the better! 🙂

failiz, 2023-09-12:
@brdane , could you specify the model formats that you would like to see? 
You mentioned STL before, but I think STEP or IGES would be better (STL is a mesh and it does not work well in CAD programs). In any case, the problem with those is that we would need a 3D model of the part and the current part format does not contain that information.  It would be possible to export the PCB without components though.

brdane, 2023-09-12:
I haven't thought of many model formats but I'm sure people have many
different preferences, mine being STL as mentioned earlier. I'm not too
sure on which ones are more often-used or standard... maybe 3DS and OBJ,
for starters? I think if there can be plugin implementation for fritzing,
that can influence a lot of programmers to make more import/export tools
that you can just drag and drop into the installation files.

KjellMorgenstern, 2023-09-17:
Please try Fritzings Gerber and IPC export together with ZofzPCB.  This will produce a 3d model of a Fritzing PCB, and it provides a basic editor to add parts that don't have a rendering. To get good results, you will need to buy ZofzPCB, but there is a free evaluation period.

failiz, 2025-11-07:
I tried to export a Fritzing project to ZofzPCB, but it did not recognized the layers automatically and it I could only see the copper. Not sure if I am doing something wrong (I have not much experienced with ZofzPCB) or if something is missing.