fritzing/fritzing-app #4061: incomplete image export SVG

bastie, 2023-09-09:
## Current Behaviour
* incomplete image export with target SVG
* PNG is complete

**Build:**
Version 1.0.1 
(rcCD-1653-0-8d480a18 2023-09-04) Cocoa [Qt 6.4.3]

**Operating System:**
macOS, M2

**Steps to reproduce:**
attachment open and export as SVG

## Expected Behaviour
Text
[issue_svg_export_problem.zip](https://github.com/fritzing/fritzing-app/files/12565902/issue_svg_export_problem.zip)

KjellMorgenstern, 2023-09-10:
The ESP part specifies that the layerid is "breadboard", but it doesn't have an id="breadboard" group in the SVG.

closed 2023-09-10 by KjellMorgenstern

bastie, 2023-09-10:
Newbie question: in result of this part is developed by https://github.com/Atsumitsu/fritzing-parts and not me, where is the id="breadboard" to insert? Can you specific this?

thx

Other question: Is it a bug or a feature to create different visual output if you export as SVG or as PNG ;-) ?

KjellMorgenstern, 2023-09-10:
Of course we could try to interpret every input the way it was likely intended, but all that guesswork will be (already is) hard to manage over time. There is a discussion in the fritzing-parts repo to improve the format and guidelines.

failiz, 2023-09-10:
@bastie , try to ask for help in the forum. There are some users really helpful and quite fast in replying.