fritzing/fritzing-app #4062: Donate page won't accept a guest login

vanepp, 2023-09-11:
## Current Behaviour

[image]

Clicking continue to Payment (which should proceed to guest checkout) loops back to the same page and doesn't allow payment. I have a paypal account but it is currently broken and I haven't bothered to try and fix it.

KjellMorgenstern, 2023-09-11:
I think we can't really debug this, since it is not happening on fritzing.org. Extract from this random google search ( https://support.thinkific.com/hc/en-us/articles/360049057454-Why-is-the-PayPal-pay-by-credit-card-option-not-being-displayed- )

**Locale/IP address**
Guest checkout is not available for all locations, as some countries are required by PayPal to have a PayPal account before making a transaction. That means that if the IP address of the buyer is determined by the system to be in a location where "PayPal account optional" is not available, they will need to create a PayPal account.
If the user is attempting to make a purchase from a location that is not their home country, they may be required to sign in to make a payment from a verified PayPal account.
**Email Address**
If the email address that the buyer is providing is linked to an existing PayPal account, the option to checkout as a guest will not be available (since PayPal has already established that the buyer has an existing account).
**Cookies**
PayPal as a business does make an effort to create new accounts. By using cookies on the user's browser, PayPal will check to see if the buyer has used the guest checkout multiple times. If so, they will remove the option and require the buyer to create a PayPal account. Since this is tracked through cookies, clearing the cache within their browser should restore the guest checkout option for the buyer.

In your case, I think it is most likely the email address is already linked to an account, or has been used several times (and PP want you to create an account now).

To mitigate this, we have added an alternative payment method:
https://fritzing.org/download?payment_method=alt

failiz, 2023-09-11:
Can we close #3865 then?

vanepp, 2023-09-11:
That looks to be the explanation for me. The alternate link worked fine and I now have 1.01 downloaded so I will close this as completed. We may want to document the alternate payment method on the donate web site for other folks with the same issue though. Thanks!

closed 2023-09-11 by vanepp