fritzing/fritzing-app #4063: LED symbol STILL incorrect

DACABBAGE, 2023-09-12:
After updating the two recent versions in June and September the LED symbol is still incorrect. In the PCB view the display shows the cathode pin (-) with the round symbol, not the square or negative symbol it should for the LED. I reported this a year or two ago when I first spotted this and was apparently missed even after two recent updates. Being just a novice  it may seem insignificant but I have to constantly remind myself when soldering an LED to flip it over to get it correct, "long wire lead goes in the square hole."  What surprise me is the fact this has not been corrected for several years or nobody else has noticed it. This can't be right? a rookie kid like me finding this and no action has been taken? Furthermore having paid for two updates I really thought something simple like this would have been fixed with two updates.

KjellMorgenstern, 2023-09-12:
Can you refer to the previous report? There might already be part of the answer to this. It is technically an easy change, but it has some implications...

DACABBAGE, 2023-09-12:
I am sorry I cannot refer to an earlier report, it was some time ago and I have no record of when it was. It could be upwards of five years ago when I first noticed this. But several address changes and new computers have left no paper trail. All I can say is I did receive a response back stating that it was incorrect and a thanks for noticing. I’ll continue putting LED’s in backwards until I hear differently! Thanks.

KjellMorgenstern, 2023-09-12:
Found it, we track it here: https://github.com/fritzing/fritzing-parts/issues/268

KjellMorgenstern, 2023-09-17:
Closing in favor of https://github.com/fritzing/fritzing-parts/issues/268

closed 2023-09-17 by KjellMorgenstern