fritzing/fritzing-app #4064: Not able to remove bins using the GUI

failiz, 2023-09-14:
## Current Behaviour
Right clik on any of the bins. The menu has options to close and delete the bin, but they are disabled, see picture.
[image]
**Build:**
Version 1.0.1 on Windows 10

## Expected Behaviour
Options are enabled and they work.

KjellMorgenstern, 2023-09-14:
Can you try importing a bin? Eg, Export the "mine" bin , then import it as "test"?

failiz, 2023-09-14:
with an imported one the options are not disabled and the action works. However, is there a reason to not allow users to remove bins from the app? In fact, you can delete the bin files from the folder where they are installed and then the bins are not shown.

KjellMorgenstern, 2023-09-14:
The general ux of handling bins could be reviewed, but i think simply enabling the delete for system bins will just cause trouble. It is difficult enough to "debug" the situation when a user has moved a bin to the end of the list and reports that it disappeared. That, and i think other issues like user bin stored in network drives or other removable drives tend to make this quite fragile.

failiz, 2023-09-14:
I checked this as a user complained in the forum: https://forum.fritzing.org/t/deleting-surplus-bins/20522
Maybe the list of active bins should be configurable in preferences. 
Note: I am not sure what closing means in this context, hides it but the parts are still loaded?

KjellMorgenstern, 2023-09-14:
It is expected behavior that deleting system bins is not possible.

closed 2023-09-14 by KjellMorgenstern

KjellMorgenstern, 2023-09-14:
Good point about "Close bin":
```
void BinManager::closeBinIn(int index) {
	if (m_stackTabWidget->count() == 1) return;

	int realIndex = index == -1? m_stackTabWidget->currentIndex(): index;
	PartsBinPaletteWidget *w = getBin(realIndex);
	if((w != nullptr) && w->beforeClosing()) {
		m_stackTabWidget->removeTab(realIndex);
		m_openedBins.remove(w->fileName());
	}
}

```
So, that should remove the bin from the view without deleting the bin file, more similar to what the user requested.

failiz, 2023-09-15:
Seems reasonable to not tackle this, but maybe we could improve the message to the user. Maybe something adding "(not allowed in core libraries)" to the options disabled?
In any case, we should create a tab in the preferences dialog to allow users to hide the bin (and maybe to specify the order). I have found several times that there are too many bins (especially when the sim bin was at the bottom). Especially as you cannot scroll the bins up and down (scrolling just goes to the next or previous bin), so it is quite tedious. Should I open an enhancement issue for that?

KjellMorgenstern, 2023-09-15:
Scrolling with the mouse wheel is on my (mental) list, there might be a ticket already, not sure. Should be easy to do now.

Ordering is already possible, just pull the bins with the mouse. Might be worth creating a short video for that, or add it to the Tip of the day.