fritzing/fritzing-app #4065: Support for More Layers in a PCB.

brdane, 2023-09-17:
## Problem
PCBs can have more than just 2 layers (front and back), and nowadays, PCBs have several. Some PCBs have up to 12, some are up to 100.

## Proposed Solution
Implement a feature to add more than two layers. Perhaps have traces color-coded. Let user be able to name layers for better organization. Modify the layer-selection button to a dropdown list of all layers, user selects which one they want to be able to modify. Change the 'View->Copper Top Layer' and 'View->Copper Bottom Layer' to a checklist that displays all layers to the user. From that point, they can check/uncheck which layers they want visible to them.

KjellMorgenstern, 2023-09-17:
This issue is not really actionable, as there are quite a bunch of open questions. Maybe a better approach would be to discuss a wishlist in the forum?

vanepp, 2023-09-18:
Right off the top more layers (without a lot of care) is going to break all the current parts which assume one or two layers. I think the appropriate answer is probably to change to one of the other EDA programs if you need more than 2 layers. I think Kicad supports more than two layers and Eagle (although it is going away soon) supports 16 layers. I expect this to be a large amount of work at the code level as it currently assumes 2 layers.