fritzing/fritzing-app #4068: The search icon is missing

ValterSartori, 2023-09-22:
## Current Behaviour
The search icon is missing. 
iMac 24 inch Apple M1 macOS Ventura 13.5.2
<please attach screenshot if possible>
**Build:**
<Versione 1.0.1  (rcCD-1653-0-8d480a18 2023-09-04) Cocoa [Qt 6.4.3]

**Operating System:**

**Steps to reproduce:**
- Step 1
- Step 2

## Expected Behaviour
Text
[image]

ValterSartori, 2023-09-23:
> Did you scroll to the end of the bins? See the small arrows in the lower
> part of the Componenti widget

I can’t see any search icon.
When I start the program is not loaded to the search icon
I need to find the Nucleo Board.
Thanks

vanepp, 2023-09-23:
Two things that may help. If your user directories are corrupted this can happen (since the search data is in there) clearing the directories may help. Note that clearing the directories will delete your mine parts bin if you have parts in it.

There are two user directories (with your parts and the parts database) which don't get touched during an install (to not affect your sketches during upgrades). On Windows they are in

c:\users\username\AppData\Fritzing\roaming\Fritzing (which is a hidden directory so you need to enable hidden directories in explorer) and

c:\Users\username\My Documents\Fritzing (where username is your windows id)

If you don't have any parts or sketches you want to keep you can just delete those two directories and Fritzing will recreate them, or you can move them aside by renaming them if you want to keep something in them. When Fritzing restarts it will recreate the directories which may cure your problem. If that doesn't work there is a reference to the needed files in the forum post below the instructions. 

linux

~/Documents/Fritzing/parts
~/.config/Fritzing 

Mac

/Users/username/Documents/Fritzing/parts
~/.config/Fritzing

https://forum.fritzing.org/t/component-search-button-missing/6496/6

KjellMorgenstern, 2023-09-25:
@ValterSartori Hai fatto scorrere fino alla fine dei contenitori? Vedi le piccole frecce nella parte inferiore del widget Componenti? I contenitori delle parti possono essere riorganizzati con il mouse, e a volte il contenitore di ricerca potrebbe semplicemente trovarsi in fondo alla lista.

ValterSartori, 2023-09-25:
Problem solved.
Dropbox made the search.fzb and my_parts.fzb files available only online
with a size of zero bytes. I made the fritzing directory available offline
and now the search icon has appeared.
Thank you, Vanepp.
Best regards

failiz, 2023-09-25:
I saw this error in the forum recently, maybe it could be a good idea to add a warning in the installer when the user selects the directory.  And ask for confirmation if someone tries to install it in a folder wich contains "Dropbox" or "OneDrive" or any other standard cloud solutions.

KjellMorgenstern, 2023-11-10:
There are two possible improvements here: 1. Add or use mechanisms to protect Fritzing files from 3rd party manipulation. We already do this by using Qt Standard File locations, but also Fritzing documents often get synced broken by such cloud solutions. I think this will not be very effectice and quiet difficult to achieve multiplatform and given the the number of variants (countless cloud solutions, network drives, removable drives) 2. Natively support the original feature in demand, sharing files in the cloud (via our own, or one that has a cross platform API for that)

Neither are withing the scope of "search icon missing" -> Closing this.

closed 2023-11-10 by KjellMorgenstern