fritzing/fritzing-app #4071: Breadboard view (and the breadboard) have a number of problems.

vanepp, 2023-09-30:
## Current Behaviour

I was testing a new power supply part when I ran in to a number of bugs relating to the breadboard. 
First by default on startup the breadboard isn't aligned to the grid and thus parts won't connect to it. It would be desirable that it start aligned to the grid (if I recall correctly this has always been the case since at least 0.9.3b.)

[image]

here we see the generic IC won't connect to the breadboard until the breadboard is moved and it aligns to the grid. Then I started testing my new part (after moving breadboard so it snapped to the grid):

[image]

all goes well here, so I switched to schematic and noticed there were no rats nest lines on the 2nd resistor. 

[image]

changing back to breadboard shows why

[image]

The breadboard layout looks completely different. Breadboard has moved up and to the left (which it shouldn't do) and failed to take the resistors with it for at least one of the resistors the connections appear to be lost. The wires have been correctly dragged to their new position still connected to the breadboard though and they come back to their correct position when the breadboard is dragged. 

[image]

dragging the breadboard back to its original position restores one resistor but not its connecting wires nor the other resistor.

[image]

clicking on the wire makes it connect (although there is a new bend point in it.

[image]

moving the other wires makes them come back (with new bend points) as well. The resistor however looks to have lost its connections although it has rats nest lines to where it should go (which didn't exist in schematic!) so it is in the routing database just not correct somehow and not correctly reflecting in to schematic (if the connections still exist!) 

[image]

rotating the resistor 90 degrees reveals the rats nest lines so the routing database appears to still know the connection is there, it just wasn't reflected in to schematic as it should have been.

[image]

Rotating the resistor 90 degrees back to its original orientation and dragging it in to position restores it.

**Build:**
Version 1.0.1 
(rcCD-1654-0-8f5f1373 2023-09-06) 64 [Qt 6.4.3]

**Operating System:**
Windows 10

**Steps to reproduce:**

See current Behaviour

## Expected Behaviour

Breadboard view should maintain its position and connections when switching between views. The breadboard should not change position or lose some of the connections. It would also be good if breadboard aligned to the grid on initial load.

failiz, 2023-09-30:
The first issue could be related to #3077
Regarding the second, could you reproduce it with core parts or just with the new power supply part? And could you reproduce it in a simpler design?

KjellMorgenstern, 2023-09-30:
I can't reproduce the breadboard issue, have never seen this except for the known (and afaik already fixed) issue with the half sized board, which was misaligned. 

It would be interesting to add the power supply part: No matter if the part is correct or has some odd errors, Fritzing should somehow gracefully work around it or reject the part.

"Jumping" parts can appear if you force delete a connection, for example: If you delete a ratsnest line in the schematic view, it means that it can not be connected in any other view anymore. For breadboards, this is done by moving the part outside the breadboard. The alternative would be to just leave them in place but disconnect (red circle instead of green). This change is too subtle, and would often be mistaken as "still connected"

vanepp, 2023-09-30:
@failiz :
> Regarding the second, could you reproduce it with core parts or just with the new power supply part? And could you reproduce it in a simpler design?

As you see below, it looks like I need to auto load the .fzpz file. If I just start Fritzing the breadboard is aligned correctly and core parts fit correctly. 

@KjellMorgenstern :
> I can't reproduce the breadboard issue, have never seen this except for the known (and afaik already fixed) issue with the half sized board, which was misaligned. When I start a new project and immediately drop a generic IC, it connects like this:

Good call, for reasons I don't understand it is the part (or the loading method)! When I start Fritzing (without loading the part) the breadboard looks to be fine:

[image]

dragging in the IC connects correctly. When I load this part by double clicking on the .fzpz file

[hanmatek-HM305.fzpz.zip](https://github.com/fritzing/fritzing-app/files/12775418/hanmatek-HM305.fzpz.zip)

(remove the trailing .zip to get the .fzpz file) 

the breadboard is misaligned. I know I have seen this before, but didn't realize it appears to be related to auto loading an .fzpz file. The power supply part is dead standard, only 4 connectors and a bunch of graphics so it really shouldn't be that. Maybe the difference in automatic loading of the .fzpz file is affecting the breadboard load? This image is from double clicking on the .fzpz file to automatically start Fritzing and load the part. As you see, the breadboard is misaligned when the load completes! Breadboard appears to be offset about 0.05in in X so the grid line is in the dead zone between pins and won't connect. 

[image]

> "Jumping" parts can appear if you force delete a connection, for example: If you delete a ratsnest line in the schematic view, it means that it can not be connected in any other view anymore. For breadboards,

That wasn't the case here, I didn't do anything but try and route the connections in schematic. It will be interesting to see if you can reproduce it with the part (which I should have thought to include in the initial report, sorry!) or if it is something to do with my machine.  I have been poking at various options:

simply start Fritzing (without autoloading the part) and breadboard is aligned as expected. 

[image]

Start Fritzing then manually load the part by File->open->fzp_file and it works fine

[image]

I thought the image was jumping (which it does somewhat probably a screen up date of the changes) and was moving but it doesn't actually appear to be. This capture is where I have clicked on the part and moved it in to breadboard but haven't yet dropped it. The ruler may (if it doesn't get moved at the same time!) if that is happening

![capture4](https://github.com/fritzing/fritzing-app/assets/16514277/8cb4b688-7900-4628-a3f6-1d13a322a1ab

but it doesn't seem to be, the breadboard is still aligned and the part looks to be in the same place I dropped it. 

[image]

so I think it looks like something about the auto loading the .fzpz file is causing the issue. It will be interesting to see if you can reproduce it with the part now I know what the sequence of events was!

failiz, 2023-10-01:
I can reproduce the misalignment of the breadboard when opening fritzing by double clicking on a fzpz file (Fritzing 1.01, Windows). I tested with five files and all triggered this bug.

KjellMorgenstern, 2023-10-02:
I can now also confirm the part is misaligned. Opening with a part skips the welcome view. When I switch to the welcome view and then back to the breadboard view, the part is correctly aligned.

KjellMorgenstern, 2023-10-02:
I found a fix, basically parts were loaded one step to late when a fzpz was given. With the fix, the breadboard is aligned correctly from start. 
I was not able to reproduce the remaining steps, both resistors had their connections indicated with ratsnest lines in the schematic.

vanepp, 2023-10-04:
I'll see if I can still reproduce it (I haven't yet gotten that far though!) the fix sounds good in any case. Thanks!

vanepp, 2023-10-08:
@KjellMorgenstern 
> I was not able to reproduce the remaining steps, both resistors had their connections indicated with ratsnest lines in the schematic.

Now after a couple of tries neither can I. Looks like something odd that I can't reproduce went on the first time (which may just be a sequence of actions I no longer remember unfortunately.) I guess we will assume your fix caught all the bugs until proven otherwise. Thanks!

vanepp, 2023-10-10:
Load the fzpz by double clicking on it (without clicking on the Welcome tab first!)  Move the breadboard to snap it to the grid. Wire the circuit in breadboard. 

[image]

switch to schematic

[image]

missing a couple of rats nest lines. 

switch back to breadboard

[image]

and the breadboard has moved and the wires look to have disconnected (as indicated by the lack of rats nest lines in schematic.)  Problem reproduced (but not usefully!) Exit Fritzing and reload the .fzpz by double clicking on it again. 

[image]

then click on the welcome tab before doing anything else. 

[image]

now switch back to breadboard where the breadboard is aligned and do the wiring. 

[image]

then switch to schematic

[image]

where all the expected rats nest lines appear. Switching back to breadboard shows no changes in position (unlike when we hadn't gone through Welcome) so it appears that going through the Welcome screen (as it should should) resolves the issue completely. 

[image]

KjellMorgenstern, 2023-10-10:
I have just pushed the possible fix, in case you want to have a look. It is still incomplete I think, seems there are more cases to take into account.

closed 2024-01-08 by KjellMorgenstern