fritzing/fritzing-app #4073: Schematic subparts appear to have a number of issues

vanepp, 2023-10-04:
## Current Behaviour

I made someone a new subpart part (detailed in this forum post

https://forum.fritzing.org/t/stereo-potentiometer/4877/7

That turns up a number of cosmetic issues (they don't appear to affect operation!) 

here is the part (delete the trailing .zip to keep github happy to get the .fzpz file) 

[Stereo-POT-subparts.fzpz.zip](https://github.com/fritzing/fritzing-app/files/12809086/Stereo-POT-subparts.fzpz.zip)

and the results

breadboard, the breadboard image is missing in the parts bin but is present in Inspector, but Inspector is missing schematic.

[image]

schematic

[image]

again a number of issues, missing icons in parts (likely identical to breadboard since it is the same image)

but all three images more or less missing in Inspector.  Moving on to the schematic image the bounding boxes for both subparts are wrong. They should be to the edge of the pins (they are in the svg!)  but instead are much wider (especially on subpart1!) and subpart1's box truncates (but doesn't block rendering as it normally would) the pin and extends well beyond the part. 

pcb 

[image]

is basically a repeat of schematic with the Inspector icons all missing identically. 

**Build:**

Version 1.0.1 
(rcCD-1654-0-8f5f1373 2023-09-06) 64 [Qt 6.4.3]

**Operating System:**

Windows 10

**Steps to reproduce:**

see Current Behaviour

## Expected Behaviour

The breadboard icon should appear in the parts bin and Inspector should have the correct images for each view. In schematic view the subpart bounding boxes should match the svg (i.e end at the end of the pins.) As noted this doesn't seem to affect operation it is all visual distraction(although the overlarge bounding box will make selection difficult when it overlaps other parts!)

failiz, 2023-10-05:
I have been playing with subparts yesterday (trying to make the oscilloscope part with two subparts in the sch view: graph and symbol). I gave up as the inspector pane does not handle well these subparts and because it was difficult to know which signal is which line in the graph if the symbol is not close to the graph (and in sch view you can have net labels).

I noticed that Fritzing makes an item for each subpart. A root item contains all subparts as children. Each subpart item has the BB and PCB view icons not set as they are in the root part. I guess that the inspector should choose the images of the root part (if it exists) instead of the ones from the child item (which are black). 

Additionally, I noticed that the properties of a capacitor part (a part using the capacitor class which was designed to allow users to change some properties, e.g., power or knob status of a potentiometer) cannot be changed in the schematic view. I suspect that the inspector pane should again show the properties of the root item instead of the subpart item or maybe change the code so the subparts are also created with a capacitor class.

vanepp, 2023-10-09:
@failiz 

> I have been playing with subparts yesterday (trying to make the oscilloscope part with two subparts in the sch view: graph and symbol). I gave up as the inspector pane does not handle well these [...]

I'm not sure how well the subpart code was ever tested. It was documented in the file format document in 2017 when we started poking at it, but I'm not sure there were any core parts that used subparts and I'm not sure how well the code was tested. As well, if you have subparts you can't have buses in the same part (I don't know why, I think I may have filed an enhancement request to fix this.) Subparts are still fairly uncommon so there may be a lot of bugs lurking. It looks like a change in 1.0.0 has affected subparts. This 74x125 part I made in 2017 

[74x125.fzpz.zip](https://github.com/fritzing/fritzing-app/files/12842023/74x125.fzpz.zip)

(strip the .zip to get the .fzpz file) 

runs correctly on  0.9.10 (other than no image in Inspector in schematic which may be due to the issues you raised above) 

[image]

In schematic the viewboxes are fine and as I would expect them to be and the entire image is reproduced.:

[image]

on 1.0.0 we have lost one of the internal lines in schematic on the part

[image]

which continues in 1.0.1 (unsurprisingly.) Unfortunately this doesn't explain another issue in schematic that I am having so I'll have to keep poking there.

KjellMorgenstern, 2024-02-03:
The subparts implementation is quite incomplete.

- parts db in some cases omits connectors from subparts (see Fritzing 1.0.2 Linux AppImage)
- part with subparts loses all connections (in the schematic) when moving from copper top to bottom in the pcb
- in some cases, switching a part e.g. from NOR to NAND, will replace one of the subparts with the complete new parts, keeping partials of the previos parts
- switching a part from top top bottom layer may create a duplicate in the schematic
- ghost connections are created in fritzing data structure when switching parts (causing invisible connections in the schematic again)
- ctrl-a, rotate, delete, undo leaves one of the subparts unrotated, and the others misaligned
- crash when selecting a subpart (not all of the subparts of one part) and then changing the variant in schematic view

closed 2024-06-26 by KjellMorgenstern