fritzing/fritzing-app #4075: Parts editor svg import on 1.0.1 is broken in at least this case.

botgeek, 2023-10-09:
## Current Behaviour
Getting an error when trying to right click on an IC image to edit a new part.

**Build:1.01
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
<Windows 10>

**Steps to reproduce:**
- Step 1 Right click on a part to edit that is a 40 pin IC

## Expected Behaviour
Expected to start editing a new part.

[image]

KjellMorgenstern, 2023-10-10:
The editor opens just fine here.

Fritzing checks for the operating systems AppDataLocation (depends on OS and OS Version). There might be multiple locations. From the possibilities, Fritzing tries to choose the first writable location.

The above error message indicates that the location used for the icon was not writable.
Unfortunately, the error message does not give the full path, only a subfolder -> this is a bug.

Please check the AppDataLocations, if there are any files, and if they are writable (user permissions?)

macOS
AppDataLocation	"~/Library/Application Support/<APPNAME>", "/Library/Application Support/<APPNAME>". "<APPDIR>/../Resources"

Windows
"C:/Users/<USER>/AppData/Roaming/<APPNAME>", "C:/ProgramData/<APPNAME>", "<APPDIR>", "<APPDIR>/data", "<APPDIR>/data/<APPNAME>"

Linux
"~/.local/share/<APPNAME>", "/usr/local/share/<APPNAME>", "/usr/share/<APPNAME>"

vanepp, 2023-10-10:
> Windows
"C:/Users/<USER>/AppData/Roaming/<APPNAME>", "C:/ProgramData/<APPNAME>", "<APPDIR>", "<APPDIR>/data", "<APPDIR>/data/<APPNAME>"

Note on Windows (which you appear to be on) AppData is a hidden directory so you need to enable Show hidden directories by clicking on the Folder options tag and selecting views (circled in green here) then click "Show hidden files and folders" (I run with this always enabled) to show Appdata in Explorer. 

[image]