fritzing/fritzing-app #4078: Build app from this repository

ashtana, 2023-10-31:
When I try build app from this repository...
I got to this error:
\fritzing-app\src\sketch\zoomablegraphicsview.cpp(89): error C2039: 'device': is not a member of 'QWheelEvent'
I to comment all code where been variable device(), but I got new fatal error LNK1104: cannot open file 'quazip1-qt5.lib'

_Originally posted by @ashtana in https://github.com/fritzing/fritzing-app/issues/4070#issuecomment-1787676983_

CrazyCoder, 2023-11-01:
[quazip_qt5.zip](https://github.com/fritzing/fritzing-app/files/13224961/quazip_qt5.zip)

Unpack this file into the folder with other dependencies (next to `fritzing-app`).

The linked video is a bit misleading and instructing to copy these files into `c:\`, they should be in `fritzing\quazip_qt5` instead.

You can inspect `fritzing-app\pri\quazipdetect.pri` file to see where it's looking for the dependencies. Also, build the release configuration of everything instead of debug. This way, you will not need to remove `d` from the end of the dll file names.

ashtana, 2023-11-01:
Thank you very much friend! Everything worked in the release64 folder, I added the DLL files and entered the command line arguments and Frizing version 1.0.0 worked for me! Thank you! This was my first experience building an application in Qt.

ashtana, 2023-11-01:
> [quazip_qt5.zip](https://github.com/fritzing/fritzing-app/files/13224961/quazip_qt5.zip)
>
> Unpack this file into the folder with other dependencies (next to `fritzing-app`).
>
> The linked video is a [...]

Thank you very much friend! Everything worked in the release64 folder, I added the DLL files and entered the command line arguments and Frizing version 1.0.0 worked for me! Thank you! This was my first experience building an application in Qt.

closed 2023-11-01 by ashtana

alastaira, 2023-11-16:
> The linked video is a bit misleading

The linked video has been removed by YouTube for apparently "violating its terms".... assuming it was a video describing how to use Qt to build an open-source project, it's hard to imagine exactly terms it breached?