fritzing/fritzing-app #4086: Please issue a tag when a new version is made public

georgesk, 2023-11-26:
Hello, I am currently maintainer of the Debian package for Fritzing. I missed some versions, because the "uscan" system which can warn me about new versions was unable to detect recent versions, like 1.0.1.

## Current Behaviour
The last tag found by "uscan" system is 0.9.6

If a git tag like 1.0.1, or v1.0.1, or V1.0.1 exists, "uscan" will automatically warn me about version 1.0.1 (uscan is based on regular expression matching and gets its information from https://api.github.com/repos/fritzing/fritzing-app/tags)

## Wished Behaviour
So please, keep somewhere a reminder for the next version change: when a new version is made public, please create a Git Tag accordingly

Thank you in advance!

GoodClover, 2023-12-24:
Void Linux has a strict policy about using versions - no arbitrary commits or betas. It seems for Arch the AUR just picked a commit that was close enough, but Void doesn't see that as acceptable.
It would be greatly appreciated if versions were tagged - the same goes for fritzing-parts!

Biswa96, 2024-01-08:
I agree with previous comment. Would it be possible to add a new tag please?