fritzing/fritzing-app #4088: Fritzing crashes on MAC

Jimmacairdrie, 2023-12-02:
Fritzing crashes, tried removing and reinstalling also tried various Fritzing versions.

## Current Behaviour
Attempts at accessing any parts result in an immediate crash or in displaying icons for the parts  then crashing. 

**Build:**
Fritzing 1.0.1 or 1.0.0 both behave similarly 

**Operating System:**
Mac Ventura 13.6.1  on an Intel i5 processor
Translated Report (Full Report Below)
-------------------------------------

Process:               Fritzing [23194]
Path:                  /Applications/Fritzing.app/Contents/MacOS/Fritzing
Identifier:            org.fritzing.Fritzing
Version:               1.0.1 (104347)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-12-02 17:42:19.4353 +0000
OS Version:            macOS 13.6.1 (22G313)
Report Version:        12
Anonymous UUID:        D5D9B39E-E7B5-BBC6-6232-B76C7D969012

Sleep/Wake UUID:       F4C6097E-5B71-443D-B3EE-16BF931FC8B9

Time Awake Since Boot: 1300000 seconds
Time Since Wake:       192137 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [23194]

VM Region Info: 0 is not in any region.  Bytes before following region: 4545204224
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      10eea5000-10f811000    [ 9648K] r-x/r-x SM=COW  ...acOS/Fritzing

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   CoreFoundation                	    0x7ff80ef282ba __CFStringCreateImmutableFunnel3 + 762
1   CoreFoundation                	    0x7ff80ef36a9f CFStringCreateWithCharacters + 38
2   Foundation                    	    0x7ff80fdf75ae +[NSString stringWithCharacters:length:] + 40
3   libqcocoa.dylib               	       0x110596b35 0x11056a000 + 183093
4   libqcocoa.dylib               	       0x1105906c7 0x11056a000 + 157383
5   libqcocoa.dylib               	       0x1105933e2 0x11056a000 + 168930
6   libqcocoa.dylib               	       0x110592783 0x11056a000 + 165763
7   QtCore                        	       0x111b415aa 0x111a84000 + 775594
8   QtGui                         	       0x10fd2e2c5 QGuiApplicationPrivate::processActivatedEvent(QWindowSystemInterfacePrivate::ActivatedWindowEvent*) + 901
9   QtGui                         	       0x10fd86238 0x10fca6000 + 918072
10  QtCore                        	       0x111b01894 QMetaMethod::invoke(QObject*, Qt::ConnectionType, QGenericReturnArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument) const + 1284
11  libqcocoa.dylib               	       0x1105b6e62 0x11056a000 + 314978
12  CoreFoundation                	    0x7ff80ef97677 __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 120
13  CoreFoundation                	    0x7ff80f0303fe ___CFXRegistrationPost_block_invoke + 88
14  CoreFoundation                	    0x7ff80f030353 _CFXRegistrationPost + 536
15  CoreFoundation                	    0x7ff80ef6a927 _CFXNotificationPost + 729
16  Foundation                    	    0x7ff80fdd6b30 -[NSNotificationCenter postNotificationName:object:userInfo:] + 82
17  AppKit                        	    0x7ff8121cdd9e -[NSWindow becomeKeyWindow] + 1359
18  AppKit                        	    0x7ff812165839 -[NSWindow _changeKeyAndMainLimitedOK:] + 995
19  AppKit                        	    0x7ff812982519 -[NSWindow _orderOut:calculatingKeyWithOptions:documentWindow:] + 1332
20  AppKit                        	    0x7ff812087f47 NSPerformVisuallyAtomicChange + 132
[... 798 lines cut ...]

      },
      "deploymentId" : 240000002
    }
  ],
  "experiments" : [

  ]
}
}

Model: iMac18,3, BootROM 522.0.0.0.0, 4 processors, Quad-Core Intel Core i5, 3.4 GHz, 8 GB, SMC 2.41f2
Graphics: Radeon Pro 570, Radeon Pro 570, PCIe, 4 GB
Display: iMac, Retina 5K (5120 x 2880), Main, MirrorOff, Online
Memory Module: BANK 0/DIMM0, 4 GB, DDR4 SO-DIMM, 2400 MHz, 0x802C, 0x344154463531323634485A2D3247334532202020
Memory Module: BANK 1/DIMM0, 4 GB, DDR4 SO-DIMM, 2400 MHz, 0x802C, 0x344154463531323634485A2D3247334532202020
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x16F), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1772.1)
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE HDD ST1000DM003, 1 TB
USB Device: USB30Bus
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
Thunderbolt Bus: iMac, Apple Inc., 41.5

SWills73, 2023-12-05:
I have the same issue and found version 0.9.10 to be stable.

Jimmacairdrie, 2023-12-05:
Many thanks for this, it works for me too!

KjellMorgenstern, 2023-12-06:
This has been reported several times, all on intel based macOS Ventura or Sonoma. So far we could not reproduce the crash, but we only have these combinations available here: Sonoma + Arm or Monterey/Big Sur/Catalina + Intel , not yet Ventura or Sonoma on Intel.

KjellMorgenstern, 2023-12-11:
I was able to try this on a fresh Ventura 13.6.1 on an Intel Xeon iMac Pro.

Fritzing 1.0.1 , downloaded from our own website. Loading several examples, as well as adding some parts to a fresh sketch (resistors, diodes, several of the ESP8266 / ESP32 ). 
Also made some traces in the PCB, added copper fill etc, and uploaded it to the Aisler Fab.

No crashes occured.

jklap, 2023-12-11:
Compiled from src (as of yesterday) for Intel Sonoma 14.1 and no crashes. I've tested creating several new sketches and even some new parts. While I know it doesn't help those that are experiencing the crashing, this is just an FYI that there are some builds that seem just fine

mxbchr, 2023-12-14:
Same thing here. Fritzing 1.0.1, macOS 13.6/Intel
Crashes shortly after startup while loading components. The time to crash can be "sped up" by searching for components immediately after the window opens. 
Downgraded to 0.9.10 and it seems to work again.

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Fritzing [23338]
Path:                  /Applications/Fritzing.app/Contents/MacOS/Fritzing
Identifier:            org.fritzing.Fritzing
Version:               1.0.1 (104347)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               502

Date/Time:             2023-12-14 18:32:04.4316 +0100
OS Version:            macOS 13.6 (22G120)
Report Version:        12
Anonymous UUID:        5E4ADFE2-3217-887F-7229-0564B8B1271C

Sleep/Wake UUID:       6774B043-B4C1-4A11-9A6C-420AE16AE4F8

Time Awake Since Boot: 1700000 seconds
Time Since Wake:       2525 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000000458a3383
Exception Codes:       0x0000000000000001, 0x00000000458a3383

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [23338]

VM Region Info: 0x458a3383 is not in any region.  Bytes before following region: 3215674493
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      105358000-105cc4000    [ 9648K] r-x/r-x SM=COW  ...acOS/Fritzing

Kernel Triage:
VM - (arg = 0x0) pmap_enter retried due to resource shortage

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   QtCore                        	       0x1080f44ca QtSharedPointer::ExternalRefCountData::getAndRef(QObject const*) + 10
1   libqcocoa.dylib               	       0x106a4d7d7 0x106a1d000 + 198615
2   libqcocoa.dylib               	       0x106a49b28 0x106a1d000 + 183080
3   libqcocoa.dylib               	       0x106a436c7 0x106a1d000 + 157383
4   libqcocoa.dylib               	       0x106a463e2 0x106a1d000 + 168930
5   libqcocoa.dylib               	       0x106a45783 0x106a1d000 + 165763
6   QtCore                        	       0x107ff45aa 0x107f37000 + 775594
7   QtGui                         	       0x1061e12c5 QGuiApplicationPrivate::processActivatedEvent(QWindowSystemInterfacePrivate::ActivatedWindowEvent*) + 901
8   QtGui                         	       0x106239238 0x106159000 + 918072
9   QtCore                        	       0x107fb4894 QMetaMethod::invoke(QObject*, Qt::ConnectionType, QGenericReturnArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument, QGenericArgument) const + 1284
10  libqcocoa.dylib               	       0x106a69e62 0x106a1d000 + 314978
11  CoreFoundation                	    0x7ff808f46687 __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 120
12  CoreFoundation                	    0x7ff808fdf40e ___CFXRegistrationPost_block_invoke + 88
13  CoreFoundation                	    0x7ff808fdf363 _CFXRegistrationPost + 536
14  CoreFoundation                	    0x7ff808f19937 _CFXNotificationPost + 729
15  Foundation                    	    0x7ff809d85b30 -[NSNotificationCenter postNotificationName:object:userInfo:] + 82
16  AppKit                        	    0x7ff80c17cd9e -[NSWindow becomeKeyWindow] + 1359
17  AppKit                        	    0x7ff80c114839 -[NSWindow _changeKeyAndMainLimitedOK:] + 995
18  AppKit                        	    0x7ff80c931519 -[NSWindow _orderOut:calculatingKeyWithOptions:documentWindow:] + 1332
19  AppKit                        	    0x7ff80c036f47 NSPerformVisuallyAtomicChange + 132
20  AppKit                        	    0x7ff80c933387 -[NSWindow _reallyDoOrderWindowOutRelativeTo:] + 618
21  AppKit                        	    0x7ff80c9337aa -[NSWindow _reallyDoOrderWindow:] + 99
[... 796 lines cut ...]
      "rolloutId" : "64c17a9925d75a7281053d4c",
      "factorPackIds" : {
        "SIRI_AUDIO_DISABLE_MEDIA_ENTITY_SYNC" : "64d29746ad29a465b3bbeace"
      },
      "deploymentId" : 240000001
    }
  ],
  "experiments" : [

  ]
}
}

Model: MacBookPro14,1, BootROM 515.0.0.0.0, 2 processors, Dual-Core Intel Core i5, 2,3 GHz, 8 GB, SMC 2.43f11
Graphics: Intel Iris Plus Graphics 640, Intel Iris Plus Graphics 640, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Memory Module: BANK 0/DIMM0, 4 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C3531324D3332443250462D3039
Memory Module: BANK 1/DIMM0, 4 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C3531324D3332443250462D3039
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x170), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1772.1)
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: AirPort, AirPort, en0
USB Device: USB30Bus
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.5
```

KjellMorgenstern, 2024-01-08:
We have released Fritzing 1.0.2 , which uses a later version of XCode, moved from Qt 6.4 to 6.5, and dropped support for macOS Catalina. Also, a possible crash due to timing issues was fixed.

Since we can´t reproduce the issue here, I can not test if Fritzing 1.0.2 fixes it or not. 

I will mark this as "Solved", and close it unless it is reported again with Fritzing 1.0.2

Jimmacairdrie, 2024-01-08:
Many thanks for this, 1.0.2 does not crash the way 1.0.1 does and so far has behaved perfectly.

SWills73, 2024-01-08:
I have also tested v1.02 and it works with none of the prior crashing issues on v1.01 with Mac OS > Nice Job! thank you.