fritzing/fritzing-app #4089: Crashes on MacOS whilst loading components

SWills73, 2023-12-04:
## Current Behaviour
Fritzing 1.01 crashes soon after loading  components.

**Build:**
Version 1.0.1 
(rcCD-1653-0-8d480a18 2023-09-04) Cocoa [Qt 6.4.3]

**Operating System:**
MacOS Sonoma Version 14.1.2

**Steps to reproduce:**
- Load Fritzing
- Click on one of the parts lists and during loading components it crashes and exits

failiz, 2023-12-11:
duplicate of #4088

KjellMorgenstern, 2023-12-11:
I was able to try this on a fresh Ventura 13.6.1 on an Intel Xeon iMac Pro.

Fritzing 1.0.1 , downloaded from our own website. Loading several examples, as well as adding some parts to a fresh sketch (resistors, diodes, several of the ESP8266 / ESP32 ).
Also made some traces in the PCB, added copper fill etc, and uploaded it to the Aisler Fab.

No crashes occured.

Tried again with Sonoma 14.1.2 and a Fritzing dev build ( 1.0.2 ).  Also no crashes.

KjellMorgenstern, 2023-12-11:
There are few reports on this crash, it seems to work for the vast majority of users.
It doesn't look like this is related to macOS on Intel CPUs, at least it works here.

What else could #4088 and #4089 have in common?

failiz, 2023-12-11:
Probably not related, but the fact that the crash is produced after dragging parts reminded  me #4025

KjellMorgenstern, 2023-12-11:
Could be. My test system for sure has different graphic and input drivers.

KjellMorgenstern, 2024-01-08:
We have released Fritzing 1.0.2, which might have fixed the problem. We can´t yet reproduce it, so I can not verify. I any case, please continue on issue #4088 .

closed 2024-01-08 by KjellMorgenstern