fritzing/fritzing-app #4093: Hover text for parts and connectors not readable in dark mode with Qt 6.5.3

KjellMorgenstern, 2023-12-14:
## Current Behaviour

[image]

**Build:**
Official builds are not affected

**Steps to reproduce:**
To reproduce:

1. Compile Fritzing with Qt 6.5.3
2. Activate dark mode
3. Let the mouse cursor hover over a part or connector

Result:
The text on the appearing note cannot be read because it has nearly the same color as the note itself.

jhstatewide, 2024-05-28:
I was able to work around this myself on KDE. Using Linux appimage:
```
export XDG_CURRENT_DESKTOP=GNOME
export QT_QPA_PLATFORMTHEME=lxqt
export GTK_THEME=Default
./fritzing-1.0.2-l1901-dbdbe34c-qt6.appimage
```

when started with the above ENV, the labels are readable and I can use fritzing normally.

KjellMorgenstern, 2024-06-26:
It should now also work with setting a theme

closed 2024-06-26 by KjellMorgenstern