fritzing/fritzing-app #4094: Pin count and labels are lost when switching package

lonetech, 2023-12-17:
## Current Behaviour
When using the Generic IC part, properties like pinout are filled in using the Inspector. 
However, the order is messed up, e.g. pin label editiing is above pin count, and changes have destructive side effects. 
Example: I've built a 74HCT00 using DIP, and want to switch package to SO. On selecting SO, existing schematic connections are broken, pin count is reset from 14 to 8, and suddenly absent (which is not what was selected) pin labels are discarded. 

**Build:**
Version 1.0.1 
(rcdebian/1.0.1-1+b1 2023-12-03) 64 [Qt 5.15.10]
**Operating System:**
Debian GNU/Linux amd64

**Steps to reproduce:**
- Add Generic IC in schematic
- Correct pin count
- Edit pin labels
- Switch package

## Expected Behaviour
Pin count should remain, leaving schematic connections unaffected. At least when an equal count is available. 

It would also be desirable to be able to change package on parts after I've saved them to my parts bin (which has to be done via Part Editor, the Add to bin... option also discards entered details). It's common to want DIP in breadboard and surface mount on PCBs. 

I also don't see how to permit multiple packages in the part editor.

KjellMorgenstern, 2023-12-23:
I have crossed out the feature requests from your report, so this issue can focus on preserving the pin labels when switching the package. This doesn't mean those ideas are rejected, they should just have separate issues with their own clear description.

KjellMorgenstern, 2025-09-25:
Bin labels are preserved. But pin count not. 

I think package switch should only be allowed to the same number of pins. 
Related to #4265