fritzing/fritzing-app #4096: Linux side scrolling no longer works on Fritzing 1.0.1

benreu, 2023-12-23:
## Current Behaviour
side scrolling on mice no longer works

**Build:**
fritzing-1.0.1-l1653-8d480a18-qt6

**Operating System:**
Linux Mint 21 Cinnamon

**Steps to reproduce:**
Push the mouse scroll wheel sideways

## Expected Behaviour
The view will scroll left and right

benreu, 2023-12-23:
Side scroll worked on 0.9.10 and no longer works in 1.0.0

KjellMorgenstern, 2023-12-23:
Which setting do you use for scrolling / zooming?

benreu, 2023-12-23:
That did it. I was using the default, and wasn't aware that could even be changed. I may have tried it way back, and forgot it again.
[image]

This setting now removes one of the gripes I had with Fritzing. That it scrolled way too slow.

[image]

Thanks for the quick response.

benreu, 2023-12-23:
Out of curiosity I went back to 0.9.10, and I see there are only 2 mouse options. 1.0.0 has 4 options.

closed 2023-12-23 by benreu

KjellMorgenstern, 2023-12-23:
We added the other two options in 1.0.0 , along with support for pinch gestures on MacBooks and other touchpads.