fritzing/fritzing-app #4098: FET p-Kanal sperrt nicht

Schildbaer, 2023-12-26:
[image]
## Current Behaviour
If Gate is HIGH Souce - Drain is open. 
<please attach screenshot if possible>
**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
<Windows 10>

**Steps to reproduce:**
- Step 1
- Step 2

## Expected Behaviour
If GATE is HIGH Sourc - Gain should be b locked

failiz, 2023-12-26:
Could you please upload the sketch?

Schildbaer, 2023-12-27:
Thank you for interested to help. 
Sorry, i can`t find a supported format to upload .fzz-file or spice-net or xml-net.
Please, advice a file-format or try to rebuild from the picture. 
Explanation of the circuit: 
The switch toggels between V+ and 0V, its common is connected to the gate of the FET.
The LED-cathode is connected to 0V. The LED-anode is connected to the FET-drain via the resistor.
The voltmeter shows the voltage at the gate. The FET-source is connected to V+. 
If gate and source on nearly the same potential the transistor shoud have an high inner resistor.
I hope, this helps 
kind regards Schildbaer

KjellMorgenstern, 2023-12-27:
.fzz files are just zip files. If github does not accept the file, you can simply rename it, like example.fz.zip

Schildbaer, 2023-12-27:
[FET p-Kanal Fehler.fzz.zip](https://github.com/fritzing/fritzing-app/files/13779133/FET.p-Kanal.Fehler.fzz.zip)

Thanks for the advice.
Kind regards Schildbaer

failiz, 2023-12-29:
I think the problem is that the VTO parameter of the model needs to be negative. You can try to find the file of the FET (basic_fet_p.fzz in core parts) and open it in a text editor. Then change the VTO model to -3.6, it should look like this:
<model>.MODEL P_MOSFET_GENERIC PMOS (LEVEL=1 VTO=-3.6 PHI=0.8 JS=0 KP=10.1)</model>
Notice that I removed "level=1 W=50u L=5u", this is optional as these parameters are not used in models and level=1 was specified twice.
After that save the file and you need to regenerate the database (Parts->Regenerate database). If everything went well, you should be able to see that the VTO parameter is negative when you select the MOSFET in the spice field of the inspector pane. And you can try to simulate and check if the behaviour is correct. 
Thanks for reporting the bug.

failiz, 2024-01-04:
If you want to try it easily I made a new part in the part editor without modifying the p-channel MOSFET part, save it, and manually change the spice. The file is below, you need to download and change the extension from zip to "fzpz". Then, you can double click on it and is imported in Fritzing:
[Basic FET P-Channel.zip](https://github.com/fritzing/fritzing-app/files/13836311/Basic.FET.P-Channel.zip)
@Schildbaer , could you try the part? I think it works well in your circuit.

@KjellMorgenstern , the new part editor does not allow to change the Spice field of the part. So, it is tricky to modify the simulation behavior, even if the users know how to do it. I had to make a new part, save it, uncompress it, modify the fzz file and compress it again. Not sure how advanced you are with the replacement of the new part editor or it if it is worth to add it to the new part editor.

KjellMorgenstern, 2024-01-05:
@failiz Indeed, we need to add the ability to try out various Spice models quickly. Parts editor or even the inspector would be good places.

failiz, 2024-10-09:
Fixed in 1.0.4: https://github.com/fritzing/fritzing-parts/commit/0537bc863586549e940557b7c0d5e1aa68ddc5f1