fritzing/fritzing-app #4100: Import netlist

Kreijstal, 2024-01-04:
## Problem
So you can have a starting template or puzzle to know what to do

## Proposed Solution
I know that a netlist wont automatically give you positions but what I mean is, the pieces of the netlist will constrain your movement, so you be forced to do anything as long as that netlist still holds, like a constrainer, and pieces are connected automatically, yes at first everything will be scrambled, messy, but the user will be able to prettify it, is that possible?

KjellMorgenstern, 2024-01-08:
There are several conditions needed for this.
1. obviously, the list loading, but that is just the tip of the iceberg
2. Right now, a connection is deleted if is not routed in at least one of the views Breadboard, Schematic, PCB.  So we would need netlists as an additional internal representation. That affects *a lot* of places in the code.
3. After we have 2., a netlist view and editor would again be not much effort.

Fritzing already has a ratsnest, dotted lines that indicate unrouted connections. So, if you created a sketch only using one of the views, for example the breadboard, you get an unrouted schematic and pcb. @Kreijstal Would loading a sketch that only has connections routed in one of the views work for you? The other view can then be used as the puzzle / template.

In general I think it would make sense to add another representation, like a Netlist view. Right now we can only export netlists in various formats, but we can not read or even edit them, like naming subnets.
A similar feature exists in other tools, for example https://forum.kicad.info/t/net-editor-tool-or-plugin/36261

A netlist loader/editor would also help with automated testing and other sorts of scripting.  We have requests from visually impaired users to load a netlist, it would allow a bigger part of the work done without vision.

Kreijstal, 2024-01-08:
Yeah that's okay for me, but maybe a netlist itself doesn't have the models themselves, so maybe there is an intermediate format, but yeah a solution ir workaround is nice, thank you

failiz, 2024-01-08:
I think this is related to https://github.com/fritzing/fritzing-app/issues/3743

Kreijstal, 2024-01-08:
> I think this is related to #3743

would be mega nice to see it in breadboard view too