fritzing/fritzing-app #4103: Can't Install 1.02 or 1.01 because of missing, unkown DLLs

Sqwarky, 2024-01-11:
## Current Behaviour
I’ve just paid for and downloaded the last two 64bit releases for Windows 11 Ver 10.0.22621 and upon failed installation, states that I don’t have required DLL files and then doesn’t bother to tell me what DLLs it needs

**Build:**
<e.g. Versionfritzing-1.0.2-l1901-dbdbe34c-win64_qt6

**Operating System:**
<Windows 11> Home

**Steps to reproduce:**
- Step 1
Scan with Bitdefender
- Step 2
Silence antivirus for 5 mins
- Step 3
double click file
- Step 4
Receive uninformative error message
[image]
- Step 5
 Exits with no log file
 
[image]

## Expected Behaviour
installation of the program or details of the failure apart from a generic message. This happens with Version 1.0.1 also.

KjellMorgenstern, 2024-01-11:
The error message is from the installer itself. The installer is the exact same as with Fritzing 1.0.1.
Further, the error message is quite typical for cases with Antivirus software blocking the installation at a step.

We have certified the binary with the official procedure from Microsoft, so Windows Defender will know about it.
You should see a certificate to "Fritzing GmbH" if you check the properties.

Although we have a certificate, it might take software like BitDefender (and all other products of this category) some time to add the new signature to their system. Fritzing 1.0.2 was released only a few days ago. We don´t really have any direct influence on BitDefender or similar software.

You can try disabling BitDefender, or simply wait another week and update it.

The DLL missing is pretty basic, there is few we (or even the company creating the installer software) can do about it.

Leaving this issue open for a while so it is easier to find.

KjellMorgenstern, 2024-01-11:
About the "missing dll". We have tested this on pretty fresh Windows installations. It seems to work for the majority of users. Together with the poor information from the error message, my guess is that the message might be misleading, and access to the DLL is blocked.

Did you try the hint from the forum?
`msiexec /i “C:\MyPackage\Example.msi” /L*V “C:\log\example.log”`

Sqwarky, 2024-01-12:
@KjellMorgenstern I disabled Bitdefender after scanning the downloaded file as detailed in Step 2, I have also ensured Windows is up to date as well.

Yes, I did try you command line hint with no success:

[image]

Paul.

KjellMorgenstern, 2024-01-12:
@Sqwarky The last error message asks to check the log file location, that would be C:\log . If you didn't change that part of the command to use an existing folder, or created this "log" folder, then the error message is expected.

Kreijstal, 2024-01-12:
Same, but antivirus disabled, also why can't it be portable, installers are kinda sucky

jfmorgenstern, 2024-01-12:
> @KjellMorgenstern I disabled Bitdefender after scanning the downloaded file as detailed in Step 2, I've even created an exception for bitdefender to ignore the file without success. Windows is up to date as well. I get the same error with Ver1.0.0 and 0.9.10 too.
> 
> Yes, I did try you command line hint with no success:
> 
> ![image](https://private-user-images.githubusercontent.com/57462285/296094120-bfc1e710-b0c0-4a06-9ac8-18d80cb7a82e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUwODIzNTMsIm5iZiI6MTcwNTA4MjA1MywicGF0aCI6Ii81NzQ2MjI4NS8yOTYwOTQxMjAtYmZjMWU3MTAtYjBjMC00YTA2LTlhYzgtMThkODBjYjdhODJlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEyVDE3NTQxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM0YjlkODU0MDgxOTU2NGFkNTM1ZTJkNzMyZWNjZjgyODAyYTA2ZjQwMTQ2NGVmNDMyMzY3M2RhMDUwNmU4NWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.tNxmLAMkYVkV0EXyiR_kCt9m_RZeKTz0UU3U-IoZeEw)
> 
> Paul.

Try to use C:\log.txt instead of C:\log\log.txt. The error indicates that the C:\log folder does not exist.

KjellMorgenstern, 2024-02-03:
No update for some time. @Sqwarky It looks like you solved it?

closed 2024-02-08 by KjellMorgenstern

Sqwarky, 2024-02-09:
Nope, I gave up on it as it wouldn't work on one desktop or two laptops.

Waste of money...

Paul.