fritzing/fritzing-app #4104: Error: C1083: Cannot open include file: 'clipper.hpp': No such file or directory

kaamburoow, 2024-01-13:
From where do I download and how should the clipper1 folder look like? I'm getting `C:\Users\Mitko\Desktop\fritzing\fritzing-app\src\svg\groundplanegenerator.h:25: error: C1083: Cannot open include file: 'clipper.hpp': No such file or directory`

jklap, 2024-01-13:
Until there is an official comment.....

Library can be found here: https://sourceforge.net/projects/polyclipping/files/

I built it using (OSX/linux):
```
# sitting in a directory at the same level as fritzing-app/
unzip clipper_ver6.4.2.zip -d Clipper1
cd Clipper1/cpp
cmake -S . -B build -D CMAKE_INSTALL_PREFIX=../6.4.2
cmake --build ./build --target install
```

kaamburoow, 2024-01-14:
@jklap I did this and now when building I get `cannot open file 'polyclipping.lib'`. The lib folder is empty and I can't find any .lib files.

closed 2024-01-14 by kaamburoow

axelklaudedelgado, 2024-01-17:
I've had the same problem

> cannot open file 'polyclipping.lib'

even after adding 
> SET(CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS ON)
in the CMakeLists.txt, cleaning, and rebuilding the project, I still have the same error.

Is the file for 'polyclipping.lib' ever created?

jklap, 2024-01-17:
It does build under OSX/Linux -- can't speak to building it on Windows.... see these files built:

```
libpolyclipping.22.0.0.dylib
libpolyclipping.22.dylib -> libpolyclipping.22.0.0.dylib
libpolyclipping.dylib -> libpolyclipping.22.dylib
```

BartmanEH, 2024-06-20:
> Until there is an official comment.....
>
> Library can be found here: https://sourceforge.net/projects/polyclipping/files/
>
> I built it using (OSX/linux):
>
> ```
> # sitting in a directory at the same [...]

I tried this but still getting 'clipper.hpp' file not found.

> > @jklap I did this and now when building I get `cannot open file 'polyclipping.lib'`. The lib folder is empty and I can't find any .lib files.
> 
> Fixed this by adding `SET(CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS ON)` in Clipper1/CMakeLists.txt.

I tried this and re-ran cmake commands but still have 'clipper.hpp' file not found.

admittedly I'm just a hack at compiling, any tips appreaciated.