fritzing/fritzing-app #4106: Debug Assertion Failed!

kaamburoow, 2024-01-19:
## Current Behaviour
It spits this error when loading a sketch: 
Debug Assertion Failed!

Program: C:\Users\Mitko\Desktop\fritzing\release64\Fritzing.exe
File: minkernel\crts\ucrt\src\appcrt\heap\debug_heap.cpp
Line: 904

Expression: _CrtIsValidHeapPointer(block)

For information on how your program can cause an assertion
failure, see the Visual C++ documentation on asserts.

(Press Retry to debug the application)
[image]

**Build:**
Version 1.0.2 (b0.9.6-1875-gdbdbe34c 2023-12-20) 64 [Qt 6.5.3]

**Operating System:**
<Windows 10>

**Steps to reproduce:**
- Build fritzing release
- Open a fritzing sketch(even example would work)

## Expected Behaviour
Open the sketch without any errors.

KjellMorgenstern, 2024-01-19:
Can you reproduce this with the official build?

kaamburoow, 2024-01-19:
I don't have the official build, but it happens with every sketch.

KjellMorgenstern, 2024-02-03:
Can not reproduce.

closed 2024-02-03 by KjellMorgenstern