fritzing/fritzing-app #4107: Quick New Part Creator/Editor

brdane, 2024-01-22:
## Problem
Existing part editor is useful, but a little difficult to navigate and start creating a foundation of new parts.

## Proposed Solution
In the PCB design window, be able to draw a silkscreen layout of part. In addition, be able to add pads, silkscreen text, and pad labels inside the outline. When complete, select everything, right-click, and have a "Make new part with selected" option be available in context menu. When selected, Fritzing automatically takes all selected things, creates a new part in the part editor, and all the user needs to do is enter in their metadata and pinouts. This proposed method of part creation will make SUCH a more convenient way to create parts.

KjellMorgenstern, 2024-02-03:
I'd suggest to rephrase this into "Create a new part from selection in PCB"

- Connectors would be automatically numbered from 1 to n.
- For connectors in the same net, a bus will be created in the part
- Wires will not be copied into the part
- Silkscreen text items will be copied into the part
- Silkscreen logo items will be copied into the part
- Holes will be copied into the part
- Vias will be copied into the part as THT connectors

Based on the number of connectors, a generic IC would be created for the breadboard view and the schematic view.
Alternative: Breadboard view could generate a breakout board with the same layout as the PCB footprint.

failiz, 2024-02-03:
It could also be that the PCB footprint modifies  an existing part (keeping PCB and schematic and replacing the PCB footprint). Thus, it will allow to modify an exiting part (new package, adding a thermal pad, increasing pads, etc.)

vanepp, 2024-02-27:
@KjellMorgenstern  

> Connectors would be automatically numbered from 1 to n.

This should be 0 to n-1 I expect.