fritzing/fritzing-app #4109: tooltip colors are so similar

andreruz, 2024-01-27:
## Current Behaviour
Tooltip colors are dificult to read. im on windows 11, background are black and font is dark gray.
[image]

**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
<Windows 10>

**Steps to reproduce:**
add any component
mouse hover any connector.

## Expected Behaviour
the tooltip is with better contrast, or any place to customize this.

KjellMorgenstern, 2024-01-27:
Version 0.9.3 is 8 years old, please try a recent version

KjellMorgenstern, 2024-01-31:
Affects tooltip of connectors

While on it, TOTD has a very dark shadow:

[image]

KjellMorgenstern, 2024-01-31:
Related #4093

tyeth, 2024-03-10:
Makes it totally unusable. Had to switch off dark mode in Windows.

closed 2024-06-26 by KjellMorgenstern