fritzing/fritzing-app #4111: Incorrect simulation of AND gate

ayusubov, 2024-02-04:
## Current Behaviour
The AND gate circuit (with three NPN BJT transistors) simulation gives wrong results. It works fine with Tinkercad for example. Could not attach the .fzz file. 
[image]
**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
<Windows 10>

**Steps to reproduce:**
- Step 1
- Step 2

## Expected Behaviour
The LED should be off in default state when both switches are released.

failiz, 2024-02-04:
Rename the fzz file to zip and upload the file. Thanks

KjellMorgenstern, 2024-02-05:
@ayusubov It would be really helpful to upload the actual sketch. The error might be in the simulator, but it might also caused by other operations, e.g. like starting with a sketch from a broken Fritzing version, or loading a incomplete Fritzing part.

failiz, 2024-02-06:
This circuit does not make sense to me. This is the schematics:
[image]

Please, try to implement another and gate or explain why this circuit should work.

ayusubov, 2024-02-07:
> This circuit does not make sense to me. This is the schematics: [...]

Thanks for the follow-up. The schematics does not make sense to me, too, as it is not the intended design. The transistors are NPN ones and the output from the serially connected two is to be connected to the base of the third, controlling switch one. Let me double-check. Strangely the "same" design works in Tinkercard and on a real breadboard.

ayusubov, 2024-02-07:
[image]
My schematics does not look as neat, sorry. Does it make more sense now?

failiz, 2024-02-16:
I think that should work. I made my version of what you tried there and it works correctly. Check it:
[and_transistors.zip](https://github.com/fritzing/fritzing-app/files/14315732/and_transistors.zip)
Note that in your sch when the switches are open, the base of the transistors is floating (in mine they are connected to the ground).
So, I do not think there is a bug in the simulator.

ayusubov, 2024-02-17:
Dear @failiz thanks a lot for the trouble. When I simulate your circuit, the LED is on by default (with both switches released), which is not right. It is off when both switches are put to the Pressed state :-(

failiz, 2024-02-18:
Yes, but the switches have pull-up resistors. Thus, when both switches are released, the input to the gate is 9V and 9V and the output is high (around 2.3V due to the LED). If you press one switch, then the input of the gate changes to 0V and the output is low (close to 0V). Thus, the 
[image]
The status of the switches does not matter at all, it is the voltage.
You could also change the circuit to pull down resistors to achive the effect that you are looking. See:
https://en.wikipedia.org/wiki/Pull-up_resistor
https://eepower.com/resistor-guide/resistor-applications/pull-up-resistor-pull-down-resistor/#
In any case, this is not a bug of the software.

ayusubov, 2024-02-18:
Let me get it straight 
- I have downloaded the file you attached, opened it with Fritzing, clicked Simulate, and it does not behave as expected (as an AND gate), in addition it shows smoke over one of the resistors. 
- The original circuit I tried to simulate works as a proper AND gate on actual breadboard, and in Tinkercard simulation, and it does not behave as expected in Fritzing.

failiz, 2024-02-18:
Let me get it straight. 
The smoke in my circuit is correct, there is too much current for a 1/4 W resistor. You are more than welcome to increase the resistance of that resistor if you want to fix it. 
The AND gate in my file works as an AND gate. The output is HIGH when both inputs are HIGH. 
I cannot debug your breadboard. There are several transistors with different properties and pinouts.
I cannot debug TinkerCAD, it is a proprietary software.
The circuit in the file you submitted does not make sense to both of us.
The picture that you submitted after that is almost identical to my circuit, but the input of the gate are floating when the switches are released, which is not a good design.
The status the switch is irrelevant to the gate. As I said, you can use pull-downs instead of pull-ups to achieve the behaviour that you want. 
Please, give a valid reason to see why my circuit is not working as expected. Truth table for my gate.

INA INB OUT
 H    H    H
 L     H    L
 H    L     L
 L     L     L

ayusubov, 2024-02-18:
Dear @failiz , the LED is on or off is the effect we are looking for. Fritzing simulation does not turn the LED on only when both switches are Pressed.

failiz, 2024-02-18:
Dear Araz,
please, change my circuit to use pull-down resistors instead of pull-up resistors for the switches. That will allow the behavior that you want. To do that, simply swap the position of the resistor and the switch for each of the inputs of the gate. 
The state of the switches determines the input signal LOW or HIGH for the gate. In my case, using pull ups, the gate input receives a HIGH signal when they are released. When pressed they receive a LOW signal. 
Best wishes!

KjellMorgenstern, 2024-02-23:
@failiz Can or should we do something about floating gates? We could detect if any of the connectors used in spice is open, and warn.

closed 2024-02-24 by KjellMorgenstern

failiz, 2024-02-26:
> @failiz Can or should we do something about floating gates? We could detect if any of the connectors used in spice is open, and warn.

Yes, we could give a warning in case of floating gate inputs or transistor bases. It could be annoying if you have a 4 gate IC and you are only using 1 of them. Not sure if the ad hoc code is worth it un less we have a way to define what is an input, etc.

KjellMorgenstern, 2024-02-26:
Right, probably not worth it for the limited use cases. It could be part of a more generic DRC improvement, like adding constraints to parts. Such a feature is currently not planned (many other improvements scheduled before that)