fritzing/fritzing-app #4123: Unable to place components in list mode

Clavdiv, 2024-03-02:
[image]
Please let me know - me - beginner level ....
Only way to capture this was with a photographic camera - mobile phone

KjellMorgenstern, 2024-03-02:
Bug template was ignored, exact version of Fritzing and also the exact OS version are missing.

KjellMorgenstern, 2024-03-02:
Can reproduce it on Ubuntu 22.04, current development branch. 
The mouse cursor in the screen photo looks different, so at least one more configuration is affected.

Clavdiv, 2024-03-03:
Peter Van Epp VANEPP suggested I forward this problem on to you - it is real and unfortunate - hope you have a fix - Fritzing ver 1.0.2 - OS Windows 10 Professional ver 10.0.1945 Build 19045 -  Cheers

Clavdiv, 2024-03-03:
Component placement would be much enhanced if working in LIST mode

closed 2024-06-26 by KjellMorgenstern