fritzing/fritzing-app #4124: Alimentation 4piles AAA

Turban-dominique, 2024-03-03:
Tension fausse
4X1.5V = 6V et non 4.8V
Dominique
Wrong voltage
4X1.5V = 6V and not 4.8V
Dominique

failiz, 2024-03-03:
It dependes on the type of chemistry used in the batteries. Alkaline batteries have a nominal voltage of 1.5, but NiMH have a nominal voltage of 1.2. 
But I agree that we should have two parts, the current one and another one with alkaline batteries. 
In any case, you can edit the part and change the voltage using a text editor.

Turban-dominique, 2024-03-04:
Bonjour,

J’ai essayé de modifier, mais c’est fixe dans une liste de 3V ; 4.2V ; 9V.

Si vous pouviez rajouter 6V cela serait bien.

Faire deux types distincts entre pile et accu et les bonnes tensions qui vont avec.

Merci de vos modifications dans les prochaines versions.

Cordialement

Dominique

 

Hello,

I tried to modify, but it's fixed in a list of 3V; 4.2V; 9V.

If you could add 6V that would be great.

Make two distinct types between battery and accumulator and the right voltages to go with them.

Thank you for your modifications in future versions.

Sincerely

Dominique