fritzing/fritzing-app #4125: Build: develop branch: ui_exportparametersdialog.h not checked in or otherwise missing

nbaileydev, 2024-03-04:
https://github.com/fritzing/fritzing-app/blob/c00d04f4e220ea24a9d53935f5c1363a24908972/src/dialogs/exportparametersdialog.cpp#L2

I'd like to help with the project, but I'm unfortunately unable to get past this include requirement.

Platform: Win11
C++: MSVC2019
Qt: 6.6.2
Boost: 1.8.4
libgit: 1.7.2
quazip: 1.4

Compile Output:
```make
19:28:17: Running steps for project phoenix...
19:28:17: Starting: "C:\Qt\6.6.2\msvc2019_64\bin\qmake.exe" C:\Users\me\src\fritzing\fritzing-app\phoenix.pro -spec win32-msvc "CONFIG+=debug" "CONFIG+=qml_debug"
Project MESSAGE: target arch: x86_64
Project MESSAGE: found libgit2 include path at C:/Users/me/src/fritzing/fritzing-app/../libgit2/include
Project MESSAGE: found libgit2 library in C:/Users/me/src/fritzing/fritzing-app/../libgit2/build64/Release
Project MESSAGE: Using fritzing boost detect script.
Project MESSAGE: using ../boost_1_84_0 boost environment
Project MESSAGE: Using fritzing ngspice detect script.
Project MESSAGE: using C:/Spice64
Project MESSAGE: Using Fritzing quazip detect script.
Project MESSAGE: found quazip in C:/Users/me/src/fritzing/quazip-6.6.2-1.4
Project MESSAGE: Using fritzing svgpp detect script.
Project MESSAGE: including C:/include
Project MESSAGE: Using fritzing Clipper 1 detect script.
Project MESSAGE: including Clipper1 library on windows
Project MESSAGE: including C:/include
Project MESSAGE: libs -LC:/Users/me/src/fritzing/fritzing-app/../libgit2/build64/Release -lgit2 -LC:/Users/me/src/fritzing/quazip-6.6.2-1.4/lib -lquazip1-qt6 -LC:/lib -lpolyclipping
Project MESSAGE: target arch: x86_64
Project MESSAGE: found libgit2 include path at C:/Users/me/src/fritzing/fritzing-app/../libgit2/include
Project MESSAGE: found libgit2 library in C:/Users/me/src/fritzing/fritzing-app/../libgit2/build64/Release
Project MESSAGE: Using fritzing boost detect script.
Project MESSAGE: using ../boost_1_84_0 boost environment
Project MESSAGE: Using fritzing ngspice detect script.
Project MESSAGE: using C:/Spice64
Project MESSAGE: Using Fritzing quazip detect script.
Project MESSAGE: found quazip in C:/Users/me/src/fritzing/quazip-6.6.2-1.4
Project MESSAGE: Using fritzing svgpp detect script.
Project MESSAGE: including C:/include
Project MESSAGE: Using fritzing Clipper 1 detect script.
Project MESSAGE: including Clipper1 library on windows
Project MESSAGE: including C:/include
Project MESSAGE: target arch: x86_64
Project MESSAGE: found libgit2 include path at C:/Users/me/src/fritzing/fritzing-app/../libgit2/include
Project MESSAGE: found libgit2 library in C:/Users/me/src/fritzing/fritzing-app/../libgit2/build64/Release
Project MESSAGE: Using fritzing boost detect script.
Project MESSAGE: using ../boost_1_84_0 boost environment
Project MESSAGE: Using fritzing ngspice detect script.
Project MESSAGE: using C:/Spice64
Project MESSAGE: Using Fritzing quazip detect script.
Project MESSAGE: found quazip in C:/Users/me/src/fritzing/quazip-6.6.2-1.4
Project MESSAGE: Using fritzing svgpp detect script.
Project MESSAGE: including C:/include
Project MESSAGE: Using fritzing Clipper 1 detect script.
Project MESSAGE: including Clipper1 library on windows
Project MESSAGE: including C:/include
19:28:22: The process "C:\Qt\6.6.2\msvc2019_64\bin\qmake.exe" exited normally.
19:28:22: Starting: "C:\Qt\Tools\QtCreator\bin\jom\jom.exe" -f C:/Users/me/src/fritzing/build-phoenix-Desktop_Qt_6_6_2_MSVC2019_64bit-Debug/Makefile qmake_all

jom 1.1.4 - empower your cores

19:28:22: The process "C:\Qt\Tools\QtCreator\bin\jom\jom.exe" exited normally.
19:28:22: Starting: "C:\Qt\Tools\QtCreator\bin\jom\jom.exe" 
	C:\Qt\Tools\QtCreator\bin\jom\jom.exe -f Makefile.Debug
Error: dependent 'C:\Users\me\src\fritzing\debug64\ui_exportparametersdialog.h' does not exist.
jom: C:\Users\me\src\fritzing\build-phoenix-Desktop_Qt_6_6_2_MSVC2019_64bit-Debug\Makefile [debug] Error 2
19:28:23: The process "C:\Qt\Tools\QtCreator\bin\jom\jom.exe" exited with code 2.
Error while building/deploying project phoenix (kit: Desktop Qt 6.6.2 MSVC2019 64bit)
When executing step "Make"
19:28:23: Elapsed time: 00:06.

```

KjellMorgenstern, 2024-03-04:
Not a bug, Fritzing builds fine, and there is no setup step or similar required to get the .ui related files. Are you already familiar with Qt? You could try some Qt examples that demonstrate .ui file use.

closed 2024-03-04 by KjellMorgenstern

nbaileydev, 2024-03-04:
I've used Qt for about 10 years, although predominantly on Linux.  The build was performed within Qt Creator 12 and none of the usual qmake/pro/pri tricks seem to allow it to generate.  There is something broken in the project.

I was actually going to help with your build process and documentation since it seems that there's a serious lack in that area.  I'm guessing the project doesn't want community contributors.  You can find some beginner & junior-level resources on the internet on managing clean, repeatable builds and cross-platform continuous integration.

MKK4559, 2024-03-11:
Try running the project using "Debug" instead of "Release" then check the debug64 folder. I found that file (along with other missing ones) there.

If that doesn't work, temporarily comment "#include "ui_fabuploaddialog.h"" (line 26) in fabuploaddialog.cpp then try it again. (I don't exactly remember how I did it.)