fritzing/fritzing-app #4126: SVG export problem - Pololu DRV 8825 not loading correctly

KotaroHarada, 2024-03-04:
## Current Behaviour
Stepper motor driver DRV 8825 not loading correctly when I export as SVG (Schematic). As BreadBoard is visivle but with white background. 
<please attach a screenshot if possible>
[image]
[image]

<please attach a Fritzing file if possible. To attach it, change the extension of the file from fzz to zip (e.g., bug.zip instead of bug.fzz)>

**Build:Version 0.9.10 
(bCD-2134-0-40d23c29 2022-07-01) 64 [Qt 5.15.2]**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
<Windows 11>

**Steps to reproduce:**
- Step 1: add drv8825
- Step 2: Export to SVG

## Expected Behaviour
DRV 8825 correctly exported as SVG

KjellMorgenstern, 2024-03-05:
Still reproducable in Fritzing 1.0.2.

Intersting, the DRV8825 part contains a photograph of the board as BASE64 encoded jpeg. That jpeg is not rendered by Fritzing, but still included in the SVG export.

A quick fix for the part is to remove the image tag:
`<image enable-background="new    "  href="data:image/jpeg;base64,/9j/4AAQSkZJRgABAgEDzgPOAAD/7AARRHVja3kAAQAEAAAAHgAA/+4AIUFkb2JlAGTAAAAAAQMAE...`

Possible fixes:
1. Fix parts that currently contain embedded images, as there are probably no parts that intentionally do this.
2. Warn on embedded images
3. Turn it into a feature. We could at least allow embedded images that have a transparency layer.

KjellMorgenstern, 2024-03-05:
Created https://github.com/fritzing/fritzing-parts/issues/392 for the 8825 part itself.

KotaroHarada, 2024-03-06:
> Still reproducable in Fritzing 1.0.2.
>
> Intersting, the DRV8825 part contains a photograph of the board as BASE64 encoded jpeg. That jpeg is not rendered by Fritzing, but still included in the SVG [...]

Can u tell me where i can change this? Im newbie and never fix anything like this in this program. I need another program to do this or something?

KjellMorgenstern, 2024-03-09:
Someone else tried a fix on the part, maybe this version works better:
https://forum.fritzing.org/t/pololu-stepper-driver-drv8825-working-pins-v3/1460/2

Please note that I have created fritzing/fritzing-parts#392 for the 8825 part itself. This issue is about Fritzing to better deal with errors in parts.