fritzing/fritzing-app #4129: Fritzing 1.0.0 and later don't render old sketches correctly.

vanepp, 2024-03-20:
## Current Behaviour

This sketch someone uploaded with parts from ~2016 no longer renders correctly in schematic in Fritzing 1.0.0 and later.

[Schematic Diagram2.fzz.zip](https://github.com/fritzing/fritzing-app/files/14661546/Schematic.Diagram2.fzz.zip)

(remove the trailing .zip to get the .fzz file) 

On Fritzing 0.9.10 it renders correctly 

[image]

starting with Fritzing 1.0.0 schematic no longer renders correclty

[image]

correcting the affected parts fixes the problem (see forum post for corrected parts.)

https://forum.fritzing.org/t/help-to-autoroute/22843/

 The difference looks to be that the parts are created by Coreldraw and something about the grouping appears to break rendering. Removing the extra group and rescaling the parts fixes them. 

**Build:**
works

Version 0.9.10 
(bCD-2134-0-40d23c29 2022-07-01) 64 [Qt 5.15.2]

breaks

rcCD-4157-0-6b14cfe9 2023-06-02) 64 [Qt 6.4.3]

and later.

**Operating System:**
Windows 10

**Steps to reproduce:**

See Current Behaviour

## Expected Behaviour

Fritzing 1.0.0 and later should render the parts as 0.9.10 does (if possible as it seem to be a QT6 issue probably) to avoid breaking older sketches and parts. This of course may not be easy ...

KjellMorgenstern, 2024-03-20:
The svg uses a resolution of almost 7 million dpi :

```
width="1.01321in"
height="1.1097in"
viewBox="0 0 6769375 7414023"
```

At 7 million dpi, a font size of 324770 seems fit. 
But that is quite an unusual resolution. 
Do you know of any tools that use that, or a large number of Fritzing parts? Also for CorelDraw, I think it needs to be explicitly configured when exporting the svg. The default would be more like 1000dpi or similar.

PS. Recent versions of Fritzing allow using 6 digit precision for gerber exports, so for the PCB, so the output format is not the limiting factor. But for schematics, I don't think it will make sense.

KjellMorgenstern, 2024-03-20:
Yes, also I tried it with a different Qt6 program, and it renders the same as in Fritzing. I guess the font size is capped, apparently around ~ 50000 dpi.

vanepp, 2024-03-23:
The issue seems a lot weirder than that. I have been experimenting with the svg with the intent of figuring out what a sensible minimum scale would be (with the intent of producing an error message for svgs with a too low scale, with a pointer to instructions to rescale the svg and rebuild the part.) It appears Inkscape is changing the viewbox values when changing from a scale of  0.000001 to 0.000002 then back to 0.000001. It does not produce the same view box that it started out with in the transition. The first svg breaks, the second (with a different viewbox value) works although the scales are both showing as 0.000001. I'll continue experimenting to try and figure out how to detect this so we could issue an error message for broken parts (as I think it unlikely we will be able to get QT to change!)

vanepp, 2024-03-24:
OK, I look to have figured out my problem. It looks like Inkscape can't do a scale more than a million to 1. The svg that breaks has a scale of 6.6 million to 1 ( calculated by dividing the viewbox values by the  height and width in the metadata.)  So one solution, (although I am not sure it is worth the effort it will take), would be to check the scale value in the svgs (perhaps just the schematic svg since it seems to be the one that is breaking but all of them won't hurt) and put up an error message pointing out the scale is too high and may not render in Fritzing with a pointer to the svg rescaling page in the forums here

https://forum.fritzing.org/t/rescale-a-svg-using-inkscape/7581

for a solution with the suggestion that the 1 to 1000 scale called for in the graphics standards is an appropriate value to change it to. I expect this will be a relatively few parts with a scale this high, and thus may not be worth doing unless it is easy, but it would help people figure out why sketches are not rendering correctly when it occurs. If it is too much work, I expect this to be rare enough that we can deal with it by people asking about it in the forums.

KjellMorgenstern, 2024-04-27:
Created issue #4138 to show a proper error message if this happens.

closed 2024-04-27 by KjellMorgenstern