fritzing/fritzing-app #4131: Don't consider obsolete parts for swapping

KjellMorgenstern, 2024-03-20:
## Problem
Obsolete parts are still offered when they have the same family and property name.
The current workaround is to rename the parts family to "[family name] // OBSOLETE".

## Proposed Solution
Fritzings sqlite database has a replacedBy field. For most parts, that field is empty, but for obsolete parts, it contains the ModuleID of the replacement parts.

When generating the property combobox, only consider parts with an empty replacedBy field.

KjellMorgenstern, 2024-03-20:
We could add a "None" magic value for replacedBy. Or we could add a improvedBy field, that could be utilized as a way to deprecate parts without directly obsoleting them. If we don't do anything about them, the trick to add an "obsolete" suffix to the family would still work.

failiz, 2024-03-20:
It would be great if a script does that (e.g. setting the replacedBy field to none, if it is empty) for all the parts in the obsolete folder.

KjellMorgenstern, 2024-03-20:
Actually that is not a lot of parts:
```
sgpt -s "find all .fzp files that do not have the text 'replacedBy', case insensitive"
find . -type f -name "*.fzp" -exec grep -iL "replacedBy" {} +
[E]xecute, [D]escribe, [A]bort: E
./Basic Force Sensing Resistor (FSR)_4.fzp
./pot_big_4.fzp
./Raspberry Pi 3.fzp
./basic_fsr_4.fzp
./kameleon.fzp
./pot_trimmer_6mm_4.fzp
./Textile Stretch Sensor_4.fzp
./alps-starter-pot9mm_5.fzp
./Dial Potentiometer with switch_4.fzp
./Textile Potentiometer_4.fzp
./pot_trimmer_12mm_4.fzp
./Arduino Nano.fzp
./Textile Analog Pressure Sensor_4.fzp
./pot-slider_4.fzp

```

KjellMorgenstern, 2024-03-20:
As a quick test, I remove the 'obsolete' from the family, so I could find the part to create a test sketch. Reloading the sketch now triggers the replacement mechanism, because there is a replacedby attribute. Of course replacement can be found for "None". Fritzing shows the error message, then leaves the sketch as is.

[image]

failiz, 2024-06-23:
Maybe this needs its own issue, but I will discuss it here to start the discussion as it is related to obsoleted parts:

I noticed that a few of my students´ reports still use the old symbol for the potentiometers. This is because they do not use the bins to find the parts, they search for potentiometer in the search menu and choose one of them. And the old obsoleted potentiometers still appear in that view (probably because they do not have the replacedBy attribute). 

Can we not show them in the search view? Should show first the regular parts and show the obsolete them in a subsection at the end? Should we use set replacedBy to None or similar?

KjellMorgenstern, 2024-06-26:
Created #4151 about the search on obsolete parts

closed 2024-06-26 by KjellMorgenstern