fritzing/fritzing-app #4136: Build: Not able to build in linux using Qt instrusctions provides

traliotube, 2024-04-27:
Build instructions extremely outdated and cannot be used even for downloading and building the source version of the App.

- First, The build instructions provided [https://github.com/fritzing/fritzing-app/wiki/1.3-Linux-notes](url) suggests to install Qt 5 libraries.
- `apt install qt5-default` does not work as the package is not even found.
- Second, using the build instructions is not working at all as the _updated_ instructions for Linux tells to use **Qt 5** but after following the listed commands word to word, the build fails as it gives an error telling to use atleast **Qt 6.5.3**.
-  This is extremely misleading as it is difficult to build from source using the provided commands. 
- Please update the instructions for using updated Qt versions and provide commands for installation that work with all distros and versions for Qt.

**Operating System:**
Ubuntu 22.04

**Steps to reproduce:**
- Step 1: Follow Build instructions

KjellMorgenstern, 2024-04-27:
Removed qt5 notes

closed 2024-04-27 by KjellMorgenstern

traliotube, 2024-04-27:
> Removed qt5 notes

So now how are we meant to install the dependencies?