fritzing/fritzing-app #4137: Build: Lack of instructions

traliotube, 2024-04-27:
> Removed qt5 notes

How are we meant to install the dependencies for building in linux now?

_Originally posted by @traliotube in https://github.com/fritzing/fritzing-app/issues/4136#issuecomment-2080398087_

closed 2024-04-27 by KjellMorgenstern

traliotube, 2024-04-27:
This project is open source just for name-sake, it is not easy for developers to build it from source without buying the paid and compiled version. This project cannot be called open-source and for developers

jklap, 2024-04-27:
> This project is open source just for name-sake, it is not easy for developers to build it from source without buying the paid and compiled version. This project cannot be called open-source and for developers

Uh, there are a lot of "open-source" projects out there MUCH worse then this.. plus I don't think there is a relationship between a project being called open-source and a requirement to being buildable unfortunately.. 

But irrespective of that statement, yes, there are some issues with the documentation on building but it is very much buildable by developers not involved with the project.

I guess a question to ask yourself if you are having issues getting it to build and are frustrated... which is cheaper... your time, or the low cost of the download? Personally, I like the challenge of getting something to build and am willing to spend that time... but for many, it may not be worth the couple of bucks for the download. Just a thought...

jklap, 2024-04-27:
> As for this, it is not a problem if those projects are meant to be modified or if its a library. But if it is a consumer end application, then I guess it needs to have a good enough documentation for building on own.

Wouldn't that be wonderful... I've seen more projects with little or no build documentation then projects with... irrespective of a library or app.. actually now that I think about it, I feel libraries have better documentation... otherwise how would one know how to use it?

Maybe this is just my approach from dealing with open source for 20+ years but I don't usually even bother with documentation. Build processes are usually what is kept up to date, not documentation. Build tools are pretty standard, Make, CMake, Gradle, sbt, etc, etc... hence understanding what dependencies are actually needed can be answered simply by looking into the build process...

```
QT_LEAST=6.5.3
QT_MOST=6.5.10
!versionAtLeast(QT_VERSION, $${QT_LEAST}):error("Use at least Qt version $${QT_LEAST}")
!versionAtMost(QT_VERSION, $${QT_MOST}):error("Use at most Qt version $${QT_MOST}")
```

The harder part, but again, poking in the build process will give you the answers, is how the dev's environment is set up. In the case of this project, the use of Clipper is a good example.. where the directory for it sits, etc.

But circling back, I guess I don't understand the, what comes across as, demand that the owners of the project fix the doco issue. It's an open source project.. you could contribute :)

I'm not trying to bait, but I'm not sure I would have spent 3 days trying to get something to build when there is a low cost download available unless I was really interested in making changes myself. And if that were the case then I'd also be interesting in improving the project overall and hence would contribute by maybe fixing up some doco ;)