fritzing/fritzing-app #4140: Family collision results in part not working

vanepp, 2024-05-08:
## Current Behaviour

Changing this fzpz file for someone breaks.

[YwRobot_Breadboard_Power_Supply_v5-broken.fzpz.zip](https://github.com/fritzing/fritzing-app/files/15254246/YwRobot_Breadboard_Power_Supply_v5-broken.fzpz.zip)

You can't drag the part in to either schematic or pcb view (if you try it comes up with a circle with a diagonal bar to indicate an error.) You can drag the part in to breadboard though and that appears to work OK.

[image]

but schematic does not. The pins and outline of the part are missing and the connections from breadboard are not appearing as rats nest lines (likely due to missing connectors!) 

[image]

pcb can't be selected or moved (I had to move pcb over the part to get this.) The rats nest lines appear

[image]

but when clicked on disappear rather than create traces. 

[image]

Changing the family name in YwRobot_Breadboard_Power_Supply_v5-broken.fzpz from "YwRobot Breadboard Power Supply v5" to "ywtest" which is the only change between YwRobot_Breadboard_Power_Supply_v5-broken.fzpz and YwRobot_Breadboard_Power_Supply_v5-works.fzpz makes the part work correctly. 

[YwRobot_Breadboard_Power_Supply_v5-works.fzpz.zip](https://github.com/fritzing/fritzing-app/files/15254520/YwRobot_Breadboard_Power_Supply_v5-works.fzpz.zip)

[image]

[image]

[image]

The problem is that I can't find the family name

"YwRobot Breadboard Power Supply v5"

in core parts. First a test of a known existing family name to make sure grep is running correctly 

owner@owner-PC /cygdrive/c/program files/Fritzing/fritzing-parts
$ grep -r "limit switch" ./*
./core/yl99_1.fzp:    <tag>limit switch</tag>
./core/yl99_1.fzp:    <property name="family">limit switch</property>
./core/yl99_1.fzp:  <description>YL-99 limit switch with pullup and LED module.</description>
grep: ./parts.db: binary file matches

which correctly finds the family name, but it finds nothing for the broken family name

owner@owner-PC /cygdrive/c/program files/Fritzing/fritzing-parts
$ grep -r "YwRobot Breadboard Power Supply v5" ./*

owner@owner-PC /cygdrive/c/program files/Fritzing/fritzing-parts

I expect Fritzing is finding a family "YwRobot Breadboard Power Supply v5" somewhere (but I don't know where!) and it causes an undetected collision and breaks the operation of the part but since the family name isn't showing up in core parts or the partsdb there may be some other problem.

**Build:**
Version 1.0.2 
(bCD-1901-0-dbdbe34c 2023-12-20) 64 [Qt 6.5.3]

**Operating System:**

Windows 10

see ## Current Behaviour

## Expected Behaviour

An error message (preferably on part load) that there is a family collision with something in core parts similar to the "there is already a part xxxx loaded in to Fritzing" or if this is something else other than a family collision (which seems unlikely to me!) correct whatever causes the problem.

vanepp, 2024-05-22:
Figured out the issue. It appears the word breadboard (it looks like any case) in the family field causes this issue (which I think is a bug!) We could fix it by banning the word breadboard in family names other than breadboards, but that is ugly in my view. The better fix would be correcting the code that doesn't appear to be finding a difference between ` <property name="family">Breadboard</property>` which is the family name for breadboards from `<property name="family">ywtest_breadboard</property>` which is the family name that breaks this part (because removing the breadboard from the family name makes the part work) although I don't know where that code is (probably in breadboard special processing somewhere though!)

KjellMorgenstern, 2025-06-24:
Still an issue in Fritzing 1.0.5