fritzing/fritzing-app #4142: expand range of electrolytic capacitor capacitance and voltage

vanepp, 2024-05-10:
## Problem
The capacitance field in Inspector will only accept 4 characters. Someone wanted a 1000uf capacitor (not an unreasonable request!) but you can't create that with this limitation. The pulldown menu is fairly limited but that doesn't really matter if you can manually set larger values, the problem is currently you can't. For instance it wouldn't be unreasonable to want a 40,000 uf capacitor but you can't currently set that. 

[image]

the most you can set is 999u (I actually typed 999uf but the 'f' appears to truncate!) 

[image]

you can set the voltage to 100V but again there are apparently only 3 characters available (I typed in 100V but only got100 in schematic) so you can't set 1000V for instance (although you could set 1KV I expect).

[image]

I note there is a range limitation that appears when you click on the capacitance value, this (and the number of characters available) are probably what wants to change. A 999,999uf top end would be better in my opinion as 40,000 uf caps are available and should be settable.

[image]

voltage has a similar limitation in number of characters and values although the listed range is much higher. Again more available characters would in my view be desirable to allow larger values. 

[image]

## Proposed Solution

Is inline with the current state above.

failiz, 2024-05-10:
Duplicate of https://github.com/fritzing/fritzing-app/issues/1943

failiz, 2024-05-10:
Workaround is to use engineering notation: e.g. 10m

vanepp, 2024-05-11:
Issue #1943 covers this better so I will close this in favor of that.

closed 2024-05-11 by vanepp