fritzing/fritzing-app #4143: 7805 regulator can't change voltages

vanepp, 2024-05-10:
## Problem
Someone wanted a 7806 regulator but you don't appear to be able to select anything except 3.3V 5V or variable in Inspector. Here I typed 6v in to the voltage field and it gets changed to variable (presumably because of the v!)

[image]

the part number will accept 7806 just fine so all that needs to change is the voltage setting. 

[image]

## Proposed Solution

I assume this property in the fzp file needs to be able to accept different voltages typed in although that may have other implications depending on what else it is used for. At present we appear to need to create new parts for each 78xx variant someone wants which won't scale well. The 7805 part doesn't appear to have spice data so that at least shouldn't be an issue. 

``
   <property name="voltage">5V</property>
``

failiz, 2024-05-10:
I checked this and the voltage regulators are PalatteItem objects. That means that their properties are fixed. Now, when you change the property, Fritzing swaps the part for another one. 
The solution would be to force to use the Capacitor class and edit the resources.xml.

The easiest workaround is to make a new part and change the property voltage to the desired one and remove the 5V from the SVG file.
 
@vanepp , regarding the SPICE models, they can be built using the properties of the part as variables. For example, all resistors use the same SPICE model and the resistance property is inserted in the SPICE model to achieve the required resistance. So, we could implement a generic SPICE model for all voltage regulators.

Another issue would be to improve the schematic symbols: Terminals with a lenght of 0.1inch, I would prefer "IN" and "OUT" instead of "voltage input", GND or ADJ should be rotated 90 degrees, etc.  
[image]

KjellMorgenstern, 2024-07-18:
A 7805 regulator with 6V would be a 7806 regulator. Basically, we want to be able to use the same symbol, footprint and breadboard view of the voltage regulator, but swap the name, label, spice, maybe pin labels, voltage property and spice model.

That could be done with an xml file, together with a template for voltage regulators , similar to capacitors or resistor templates.