fritzing/fritzing-app #4144: Resizing the schematic frame

Danebae, 2024-05-10:
## Current Behaviour vs Previous Behaviour (Expected )
In version 0.9.6, if I change the size of the frame, the box with the information remains the same size and the external rectangle changes size so I can include the entire schema inside the rectangle.

In version 1.0.2 it seems the external rectangle has been eliminated. Changing the size modifies the size of the box. There is no the external rectangle any more.

**Build:** 
1.0.2

**Operating System:** 
Windows 10

**Steps to reproduce:**
- Step 1: Include a schematic frame from core parts
- Step 2: Resize the frame

**Behaviour 0.9.6**
[image]
[image]

**Behaviour 1.0.2**
[image]
[image]

leiwangphd, 2024-10-27:
any update on this bug?

Danebae, 2025-09-10:
any update on this bug?