fritzing/fritzing-app #4145: La carga falló con el código de salida -1, 0

ILordC, 2024-05-13:
Es primera vez usando este programa fritzing y quería simular el Arduino para encender un let pero al momento d ejecutar el código me sale el siguiente error. "Ejecutando D:/Programas/Arduino IDE/Arduino IDE.exe --board arduino:avr:uno --port \\.\COM4 --upload C:\Users\Usuario\Downloads\prueba xxxx_TMP\prueba xxxx_TMP.ino
La carga falló con el código de salida -1, 0". No se a que se deba, Acepto todo tipo de ayuda .
[image]

failiz, 2024-05-13:
Duplicate of #4010.

Usa Arduino 1.8 en vez de arduino 2.x. Si estas en Windows, selectiona arduino_debug.exe mejor que arduino.exe. Mira  #4011

KjellMorgenstern, 2024-05-13:
Closing in favor of #4010

closed 2024-05-13 by KjellMorgenstern